mikakaraila/node-red-contrib-opcua #364: Configuring an OPC UA Client node via a json file

MaithiliDeshmukh, 2021-12-08:
Hello,
I am trying to parse a test_config.json file to the OPC UA Client node. My prosys server has only one endpoint with Basic256Sha256 and SignAndEncrypt security. When there is no security policy and security mode, my program works fine. But in this case, I get the error shown in the screenshot. I don't understand where I am going wrong. Can you please point it out?
Thanks,
Maithili
[test_config.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7675520/test_config.txt)
[image]
[image]
[flows.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7675536/flows.txt)
[](url)

MaithiliDeshmukh, 2021-12-08:
PS: I have purposely kept only this one endpoint in Prosys.

mikakaraila, 2021-12-08:
You have to provide also username & password as you use Sign & Encrypt.

MaithiliDeshmukh, 2021-12-08:
I have included username and password. It still doesn't work..

mikakaraila, 2021-12-08:
Yes, I found out reason. Only endpoint can be changed by msg.
I will fix this now.

mikakaraila, 2021-12-08:
Actual error was inside function node as you parse parameters from file and set them to msg object it will remove topic and other properties.

First set msg and then other properties fixed flow:

[image]

MaithiliDeshmukh, 2021-12-09:
Unfortunately it still shows the same error.. Could you please share your flow.txt if it is working for you?

mikakaraila, 2021-12-09:
Here: 
[ONE-ENDPOINT.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7685412/ONE-ENDPOINT.txt)
[test_config.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7685416/test_config.txt)

[image]

[image]

mikakaraila, 2021-12-09:
I fixed also lower part of the flow to use readmultiple, topic items you had in your test_config.json
[ONE-ENDPOINT.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7685702/ONE-ENDPOINT.txt)

[image]

MaithiliDeshmukh, 2021-12-10:
Hi, in the prosys server, how many endpoints do you see at your end? I still get the same error.
[image]
[image]
[image]

mikakaraila, 2021-12-10:
Ack, I added some more verbose and username & password was NOT updated from the JSON file.

BUT strange is that even user Identity is now updated correctly it still gives same error. I have to look how node-opcua checks endpoints.

mikakaraila, 2021-12-10:
I forgot to restart server as I tested this. Have to rest a little bit... I will continue tomorrow.

mikakaraila, 2021-12-11:
Parameters are now correct, but still invalid endpoint. I expect this has something to do with Prosys server, I will study more.

[image]

MaithiliDeshmukh, 2021-12-11:
I tested this program with a siemens opc ua server too. I get "null" values, which means that the connection is not established, I think. The endpoint from the json point gets parsed correctly, but all other parameters like security mode, security mode and login- dont seem to change the client node.

mikakaraila, 2021-12-11:
Yes, I have fixed those this way, 102-opcuaclient.js:
[image]

        opcuaEndpoint = {}; // Clear
        opcuaEndpoint = msg.OpcUaEndpoint; // Check all parameters!
        connectionOption.securityPolicy = opcua.SecurityPolicy[opcuaEndpoint.securityPolicy]; // || opcua.SecurityPolicy.None;
        connectionOption.securityMode = opcua.MessageSecurityMode[opcuaEndpoint.securityMode]; // || opcua.MessageSecurityMode.None;
        verbose_log("NEW connectionOption security parameters, policy: " + connectionOption.securityPolicy + " mode: " + connectionOption.securityMode);
        if (opcuaEndpoint.login === true) {
          userIdentity.userName = opcuaEndpoint.user;
          userIdentity.password = opcuaEndpoint.password;
          userIdentity.type = opcua.UserTokenType.UserName;
          verbose_log("NEW UserIdentity: " + JSON.stringify(userIdentity));
        }
        verbose_log("Using new endpoint:" + stringify(opcuaEndpoint));

mikakaraila, 2021-12-11:
There is something else than injecting parameters. I tested same parameters inside Client node and it cannot connect.
Same error: Invalid endpoint...
[image]

It will help if you can test this with Siemens OPC UA Server.

mikakaraila, 2021-12-11:
I think I found out reason finally, node-red client certificate was not trusted on Prosys Simulation server.
I will commit and push my code changes.

[image]

MaithiliDeshmukh, 2021-12-11:
My certificate is trusted. How can I make the changes you have shown in 102-opcuaclient.js? Or is there something else I need to do now?
[image]

Thank you for your help!!

mikakaraila, 2021-12-11:
I already published new version, install it.

MaithiliDeshmukh, 2021-12-15:
Thank you. It works now! :)

closed 2021-12-15 by MaithiliDeshmukh