mikakaraila/node-red-contrib-opcua #365: OpcUa - Method node continuously making new sessions

schaffner, 2021-12-10:
Hello everybody,
I am unsure if this is an actual issue or intended behavior., but it seemed weird to me.

When using the OpcUa-Method node in a flow it will connect to the Opc server and create a session. So far so good.
But since this session has  the default SessionTimeout of 60000 it expires pretty much every minute it has no activity. The result of this is that every minute a new sessions gets created on the server. 
So far this hasn't caused any major issues(only 2 times an error Bad_SessionNotActivated, but I am unsure if it was really caused by this), but in my opinions this is not ideal. 

Would it be possible to keep the first created session alive by having some kind of heartbeat signal?

[image]
[image]
[image]
l

mikakaraila, 2021-12-10:
Ok I expect it is not using keepalive. But it should open session only to execute method. I will refactor node at some time

mikakaraila, 2021-12-11:
Ok, now session will be opened as method is executed:

[image]

mikakaraila, 2021-12-11:
I added some error handling and session is closed after method is executed.

[image]

mikakaraila, 2021-12-12:
Published v0.2.250

AMBFrancesco, 2021-12-21:
Hello there,

Maybe it's me, and I don't know if it's the right section to ask, but I can't pass values from msg.payload to the method node. 
I have a method that multiplies a & b and give the result. 

With plain text numbers inserted in directly in the node it works, when I try to pass values from msg it returns 'NaN'.

I'm passing a inject node with msg.payload[0] and msg.payload[1] with 2 floats, I tried with several notations in the value fields:

- {{payload[0]}}, {{payload[1]}}
- {payload[0]}, {payload[1]}
- {msg.payload[0]}, {msg.payload[1]}
- msg.payload[0], msg.payload[1]
- payload[0], payload[1]

What's the right sintax? (Is there any documentation about it somewhere?) thanks in advance

closed 2022-01-05 by mikakaraila