mikakaraila/node-red-contrib-opcua #366: Client node, new actions: Read file & Write file

mikakaraila, 2021-12-12:
Added new functionality:

[image]

mikakaraila, 2021-12-13:
I published this yesterday v0.2.251

mikakaraila, 2021-12-13:
Added File support to server node:

[image]

MaithiliDeshmukh, 2021-12-15:
Hi Mika,

I want to use the CopyFileToServer method shown in UAExpert. How can I program this method in Node Red? 
My target file name is 'Sinumerik/FileSystem/Part Program/VCS1_1.spf' , the file is a sub program called VCS1_1.spf. Should I use the OPC UA Method node? I am not sure how to enter the object id, method id and arguments.

Would appreciate your help. Thanks!

[image]

mikakaraila, 2021-12-20:
UaExpert has support for the UAFile object:
[image]

mikakaraila, 2022-01-05:
Look FreeOpcua and asyncua Python client can also read/write file.

closed 2022-01-05 by mikakaraila