mikakaraila/node-red-contrib-opcua #367: Add multiple items to a subscription in one shot ?

DamienInfoIndus, 2021-12-16:
Hello, 
Is it possible to add multiple items to a subscription in one shot ? 
I tried to pass the nodeId and datatype of my items by array but didn't succed.

I have this warning message when nodeRed start in the prompt, is it bad ?

[image]

Thanks in advance.

mikakaraila, 2021-12-16:
No, it is not supported. This will need small change into the code.

mikakaraila, 2021-12-17:
Could you provide example? Just to make sure that I make modification so it will work with you idea/need.

DamienInfoIndus, 2021-12-17:
Yep, here !

[image]

mikakaraila, 2021-12-17:
Function MakeArrayItemsOPCUAtoSub is the one I need to know, how you have put each item into the array.

DamienInfoIndus, 2021-12-17:
Oh, something like this. 
[MakeArrayItemsOPCUAtoSub.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7735117/MakeArrayItemsOPCUAtoSub.txt)

[image]

mikakaraila, 2021-12-17:
Thanks, now this is clear. I will implement this.

mikakaraila, 2021-12-17:
OK, seems to work. I will publish this after some more testing. Tomorrow or on Sunday.

[image]

mikakaraila, 2021-12-18:
Published v0.2.253 and examples contains OPCUA-MULTI-SUB.json

DamienInfoIndus, 2021-12-20:
Thanks for your time! 
However it's seem to have some bugs : 
1. When I create the subscription with 3 differents items, I receive 3 time the same values of the LAST item of the array used to create the sub.
2. When I change the value in the same time for the 3 items, it does the same. 
3. When I change the value of each item one by one, works fine.

[image]

mikakaraila, 2021-12-20:
Ack, have to take a look into the code.

mikakaraila, 2021-12-20:
OK, small change and I modified flow, it doesn´t need datatype:
[image]

I will publish fix next.

mikakaraila, 2021-12-20:
Published v0.2.254

DamienInfoIndus, 2021-12-22:
Hello, 
Just test it on the new version 0.2.254 and run into some troubles,
on deployment and on update multiple subscribed items at the same time.

When I change the value of each item one by one, works fine.

I've attached a picture of the issue.

[image]

mikakaraila, 2021-12-22:
I expect it will initialize values always as subscription is created. It is default action and most probably according the standard.

I tested flow with Prosys UA Simulation Server with simulation variables and static variables.

mikakaraila, 2021-12-27:
Your function node causes value to be undefined

mikakaraila, 2022-01-05:
No more comments? You can reopen if there is still something.

closed 2022-01-05 by mikakaraila

Eldarion85, 2022-02-04:
Hi, overall it works fine with the "multiple subscription". But I have two questions about this.

1.
If I subscribe to an item in the normal way, information is displayed in the debug. It contains, among other things, the interval I have set.
[image]

If I subscribe to several variables via an array, I can specify an interval, but it's not used. Also, I don't know if I'm specifying this parantally for all items or separately for each item in the payload array.
[image]

2.
The second problem is the same as explained above. I have described it again in my screenshot.
After the first run everything works fine and the right items with the right values are always displayed.

> 1. When I create the subscription with 3 differents items, I receive 3 time the same values of the LAST item of the array used to create the sub.
> 2. When I change the value in the same time for the 3 items, it does the same.
> 3. When I change the value of each item one by one, works fine.