mikakaraila/node-red-contrib-opcua #368: Uncaught Exception - OpcUa-Server node

javor55, 2021-12-17:
Hello,
I have a newly installed red node on a Windows computer, when I add a server node, it crashes, as you can see in output. The reason may be in the output, but I can't find it. Do you have any idea where the problem might be?

Thank you

```console
C:\Users\demo>node-red
17 Dec 11:24:12 - [info]

Welcome to Node-RED
===================

17 Dec 11:24:12 - [info] Node-RED version: v2.1.4
17 Dec 11:24:12 - [info] Node.js  version: v16.13.1
17 Dec 11:24:12 - [info] Windows_NT 10.0.19041 x64 LE
17 Dec 11:24:14 - [info] Loading palette nodes
17 Dec 11:24:19 - [info] Dashboard version 3.1.2 started at /ui
17 Dec 11:24:19 - [info] Settings file  : C:\Users\demo\.node-red\settings.js
17 Dec 11:24:19 - [info] Context store  : 'default' [module=memory]
17 Dec 11:24:19 - [info] User directory : \Users\demo\.node-red
17 Dec 11:24:19 - [warn] Projects disabled : editorTheme.projects.enabled=false
17 Dec 11:24:19 - [info] Flows file     : \Users\demo\.node-red\flows.json
17 Dec 11:24:19 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

17 Dec 11:24:19 - [info] +-----------------------------------------------------
17 Dec 11:24:19 - [info] | uibuilder v4.1.4 initialised
17 Dec 11:24:19 - [info] | root folder: \Users\demo\.node-red\uibuilder
17 Dec 11:24:19 - [info] | Using Node-RED's webserver at:
17 Dec 11:24:19 - [info] |   http://192.168.20.159:1880/ or http://localhost:1880/
17 Dec 11:24:19 - [info] | Installed packages:
17 Dec 11:24:19 - [info] |   jquery, socket.io
17 Dec 11:24:19 - [info] +-----------------------------------------------------
17 Dec 11:24:19 - [info] Server now running at http://127.0.0.1:1880/
17 Dec 11:24:19 - [info] Starting flows
17 Dec 11:24:19 - [info] Started flows
17 Dec 11:24:19 - [red] Uncaught Exception:
17 Dec 11:24:19 - [error] Error
    at assert (C:\Users\demo\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-assert\source\index.ts:12:21)
    at ReaderState.finish [as _finish] (C:\Users\demo\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-address-space\source\loader\load_nodeset2.ts:878:31)
    at ReaderState._on_finish (C:\Users\demo\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-xml2json\source\xml2json.ts:163:18)
    at ReaderState._on_endElement (C:\Users\demo\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-xml2json\source\xml2json.ts:213:18)
    at SaxLtx.<anonymous> (C:\Users\demo\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-xml2json\source\xml2json.ts:366:36)
    at SaxLtx.emit (node:events:390:28)
    at SaxLtx._handleTagOpening (C:\Users\demo\.node-red\node_modules\node-red-contrib-opcua\node_modules\ltx\lib\parsers\ltx.js:44:14)
    at SaxLtx.write (C:\Users\demo\.node-red\node_modules\node-red-contrib-opcua\node_modules\ltx\lib\parsers\ltx.js:192:20)
    at Xml2Json.parseString (C:\Users\demo\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-xml2json\source\xml2json.ts:316:16)
    at Xml2Json.parseString (C:\Users\demo\.node-red\node_modules\thenify\index.js:65:46)

C:\Users\demo>
```

mikakaraila, 2021-12-17:
It cannot load nodeset, it is missing or you are using your own nodeset?

javor55, 2021-12-17:
I did't change anything in setting.js so it should use default folder with nodes. All is instaled via user settings - palette - install. Node-RED was installed yesterday with `npm install -g --unsafe-perm node-red` command

mikakaraila, 2021-12-17:
Node-RED installation doesn´t contain node-red-contrib-opcua installation nodesets (xml files).

Windows environment:

[image]

javor55, 2021-12-17:
I'm probably not such an advanced user, but what is the solution? I also have the files I see in your screenshot in my environment.

mikakaraila, 2021-12-17:
Or test environment with command:
node-red -v test.json

It will show a bit more.

javor55, 2021-12-20:
There is console output with `node-red -v test.json`. I instald it with palette manager.

[image]

mikakaraila, 2021-12-20:
OK first check node-opcua version, command:
npm ls node-opcua

javor55, 2021-12-21:
[image]

mikakaraila, 2021-12-21:
OK, I have to update mine, if there is somethings changed with the nodesets (most probably XML files need to be updated),

mikakaraila, 2021-12-21:
Tested and work, it must be something else broken in your environment. Did you try to reinstall node-red-contrib-opcua?

C:\Users\karaimi\.node-red>npm ls node-opcua
node-red-project@0.0.1 C:\Users\karaimi\.node-red
+-- node-opcua@2.62.7
`-- node-red-contrib-opcua@0.2.254
  `-- node-opcua@2.62.7 deduped

mikakaraila, 2022-01-20:
If no comments or updates, I will close this now.

closed 2022-01-20 by mikakaraila