mikakaraila/node-red-contrib-opcua #369: Folder for PEM

Sailman83, 2021-12-19:
Hello,

I use this nods in Docker Node-RED.

I would like know where is stocked the PEM file to make Persistent data.

Thanks for your help

mikakaraila, 2021-12-20:
Start with option -v verbose then you can check it from the connection options. 
Normally it is under current user home directory.

Connection options:[{"securityPolicy":"1","securityMode":1,"clientCertificateManager":"2","requestedSessionTimeout":300000,"clientName":"3","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"4","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None",{"untrustUnknownCertificate":true,"state":0,"folderPoolingInterval":5000,"_watchers":"5","_readCertificatesCalled":false,"_filenameToHash":"6","_thumbs":"7","_pending_crl_to_process":0,"queue":"8","location":"9","keySize":2048,"referenceCounter":0,"automaticallyAcceptUnknownCertificate":true},"",{"maxRetry":10512000,"initialDelay":5000,"maxDelay":30000},[],{},{"rejected":"10","trusted":"11","issuers":"12","crl":"13","issuersCrl":"14"},[],"**C:/Users/karaimi/AppData/Roaming/node-red-opcua-nodejs/Config/PKI**",{},{},{"certs":"15"},{},{},{}]

Sailman83, 2021-12-20:
Hello,

Thanks for your answer. But I'm not on Windows, it's containeron Linux.

I have already try with "Debug" but the "OPC UA clien " sen nothing.

I don't know here you activate "Verbose" on Nodered

mikakaraila, 2021-12-20:
From the command line, haven´t checked how to do it with Docker container.
node-red -v test.json

mikakaraila, 2021-12-20:
Check this:
[image]

Sailman83, 2021-12-20:
I thinks the folder is wrong

[image]

Sailman83, 2021-12-20:
[SOLUTION FIND]

I have create folder in persistent volume
In container I have copied the *pem file in persitent volume
After in Node-red I have indiqued the file *pem

I use always same certificat for my OPC-UA server

Thanks for your time

closed 2021-12-20 by Sailman83