mikakaraila/node-red-contrib-opcua #370: TypeError: async.parallelLimit is not a function

ashok0617, 2022-01-08:
@mikakaraila I am getting typeerror in OPCUACertificateManager._readCertificates as shown below 
[image]

I am getting this error on my k8s pod container with new images only. The same code was with node-red-contrib-opcua v0.2.240 with docker image created in November. When I build image now with same code throwing the attached error - no matter what I did to resolve. I need to address production issue and need your help. Attached pod log.
[logs-from-ifl-opc-adapter-1-637671383321936361-in-ifl-opc-adapter-1-637671383321936361-6bd96959fc-2vnrt.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7832535/logs-from-ifl-opc-adapter-1-637671383321936361-in-ifl-opc-adapter-1-637671383321936361-6bd96959fc-2vnrt.txt)

Note: Hope you remember, we work together to fix subscription issue on flow start when OPC server became available later.

mikakaraila, 2022-01-08:
It is somehow inside node-opcua-pki that uses async parallelLimit feature. Have you checked async version?
And of course node-opcua-pki version!

ashok0617, 2022-01-08:
> It is somehow inside node-opcua-pki that uses async parallelLimit feature. Have you checked async version? And of course node-opcua-pki version!

Below are version details:
async: "_resolved": "https://registry.npmjs.org/async/-/async-0.1.22.tgz",
node-opcua-pki: "_resolved": "https://registry.npmjs.org/node-opcua-pki/-/node-opcua-pki-2.12.1.tgz"

I am not sure what was in working old docker image.

> 
> There is error before async error: Cannot find private key /usr/src/node-red/.config/node-red-opcua-nodejs/PKI/own/private/private_key.pem'
> 
> Check your certificates!
> 
Had problem on flow start where OPCUA node tries to create self-signed certificate. So very first execution fails always as OpcUa-Client fails to proceed even after creating certificate. So flow looking for certificate error then subscribing the node. This is done because there is no way to restart flow on same container to continue after certificate created. This is working in production with old docker image created in oct/nov 21.
This is another issue with OpcUA-Client, looks timing issue where even node creates certificate fails to continue. Possibly this is not the cause as the same flow in production and working all the time.
[image]

> Your flow seems to use Azure AMQP and it has also dependency to async? Could it cause async version mismatch?
Yes. Using AMQP node for local (Not Azure) RMQ connection which seems to be OK
"@meowwolf/node-red-contrib-amqp": "^2.4.0"
[image]

ashok0617, 2022-01-09:
> It is somehow inside node-opcua-pki that uses async parallelLimit feature. Have you checked async version? And of course node-opcua-pki version!
> 
@mikakaraila , Found that async 0.1.22 was being used by default. Not sure why latest version of this package 3.2.2 not applied on <npm install>. I installed async package on top of <npm install> and now recheck that this package applied. Based on this change new docker image also started working. Going to rollout in production on Monday. By the way opcua package is dependent such dependencies then how come we make sure right version of package applied?

mikakaraila, 2022-01-09:
OK, good that you found it.

closed 2022-01-11 by ashok0617