mikakaraila/node-red-contrib-opcua #373: Subscribe to "custom events"

ostling, 2022-01-17:
Hi,
This is not an issue - only a question.
Thanks so much for creating this component!

I am trying to subscribe to a "custom event" from a label printer called "IOStatusChangeEventType" and this event has a property called "Status" of type UInt32 which contains the intresting information.
The event as i understand it is a subtype of the "BaseEventType" (?).
The vendor specifies this:
[image]
https://www.cab.de/en/news/news/opcua/

Browsing in UAExpert I see the event type here:
[image]

And the event is generated by this node:
[image]

So creating a event subscription for BaseEventType on root node "ns=3;i=10006" is working to get the events triggered correctly but I cannot figure out how to get the extra information field called "Status":
[image]

I have tried just pushing the status to the selectClause and eventsFields structures like so:
[image]

but then I get this error leading me to think I need to specify the custom event somewhere so it can be encoded:
D:\path\to\app\node_modules\node-opcua-types\dist\_generated_opcua_types.js:10758
        (0, node_opcua_basic_types_1.encodeArray)(this.selectClauses, stream, (obj, stream1) => { obj.encode(stream1); });
**TypeError: obj.encode is not a function**

I would appreciate any help greatly! I also aplogize if this is the wrong place to state such a question.
Best regards,
Martin

mikakaraila, 2022-01-17:
Code generates basic event fields, look this, perhaps if you want to try add it here:
[image]

Then code should "provide" you basic fields:
[image]

ostling, 2022-01-17:
Thank you so much @mikakaraila - it worked like a charm :)
So now I understand why my direct modifications to the object structure would not work - since the objects returned from the constructEventFilter is wrapped around some prototypes (which is not shown by node red debugger but clearly visible when logging the structure):
[image]

So since I dont want a customized node-red-contrib-opcua for now I just imported node-opcua via functionGlobalContext in NR and added my property via function node as such:
[image]

> let myEventFields = "3:Status"
> 
> var opcua = context.global.get('opcua');
> var myEventFilter = opcua.constructEventFilter(myEventFields);
> msg.eventFilter.selectClauses = msg.eventFilter.selectClauses.concat(myEventFilter.selectClauses)
> msg.eventFields = msg.eventFields.concat(myEventFields);
> 
> return msg;

Many thanks again!

closed 2022-01-17 by ostling

yayitazale, 2023-03-06:
> Thank you so much @mikakaraila - it worked like a charm :) So now I understand why my direct modifications to the object structure would not work - since the objects returned from the [...]

Hi:

I'm how can I add more than 1 event fields to your fuction? I'm abble to read one but I don't see how I can define more than 1 event fields for the same event type.

Thanks!

simon0168, 2023-05-05:
@yayitazale Hello I also work with an opc UA E77 server, I have the error: "TypeError: Cannot read properties of undefined (reading 'constructEventFilter')".
I think it appens because I didn't imported node-opcua via functionGlobalContext in NR.. but i dont know how to do that.

yayitazale, 2023-05-05:
> @yayitazale Hello I also work with an opc UA E77 server, I have the error: "TypeError: Cannot read properties of undefined (reading 'constructEventFilter')". I think it appens because I didn't imported node-opcua via functionGlobalContext in NR.. but i dont know how to do that.

You need to edit the **setting.js** file of the node-red and change:

```
functionGlobalContext: {
}
```

to

```
functionGlobalContext: {
opcua:require('node-opcua')
}
```

simon0168, 2023-05-05:
Thanks for your reply @yayitazale , it solved my problem! But now I have another error "TypeError: Cannot read properties of undefined (reading 'selectClauses')"

yayitazale, 2023-05-05:
> Thanks for your reply @yayitazale , it solved my problem! But now I have another error "TypeError: Cannot read properties of undefined (reading 'selectClauses')"

Sorry I forgot to tell you that you need to do a `npm install node-opcua` from the path where the settings file is located.

simon0168, 2023-05-05:
Hello, now i have "TypeError: Cannot **set** properties of undefined (setting 'selectClauses')"
Have you something in your inject node? Or it's a basic timestamp?

Koseng, 2023-05-05:
Hi guys,
don't know exactly what you are doing, but meanwhile there is support for custom events out of the box. You will automatically get all fields of the event. (it worked some time ago, did not recheck).

[image]

Just click custom event, and add the node-id of the event-type. For standard opc ua events i=xx is enough for real custom one you need the namespace as well. Like for example ns=2;i=55.

yayitazale, 2023-05-09:
> Thanks for helping @yayitazale. Now i have "TypeError: Cannot **set** properties of undefined (setting 'selectClauses')" Have you something in your inject node? Or it's a basic timestamp?

No, just a timestamp, but you must fill the OPCUA event node with correct data.

> Hi guys, don't know exactly what you are doing, but meanwhile there is support for custom events out of the box. You will automatically get all fields of the event. (it worked some time ago, did not [...]

Ok thank you, I'll have a look but we are using a custom build of nodered so in this moment we are still on node 14.18 in our IoT devices so I can't try this easily.