mikakaraila/node-red-contrib-opcua #374: Save / Load server address space

mikakaraila, 2022-01-18:
New feature to save / load address space. Loading address space will require server restart.

[image]

mikakaraila, 2022-01-18:
Published v0.2.256 and tested with Siemens nodeset.xml

anxuae, 2022-01-22:
Hi @mikakaraila, it seems a great features!
Is it possible to save it has a JSON file which is better integrated to the node-Red environment than XML (and JavaScript language)?

mikakaraila, 2022-01-23:
OPC UA specification is for xml. All programs are using xml. Siemens export is xml format etc.

https://reference.opcfoundation.org/v104/Core/docs/Part6/B/

mikakaraila, 2022-01-27:
OK, we need to wait fix for node-opcua. XML file with nodeId with " characters seems to cause problems.

mikakaraila, 2022-01-27:
Ok, more investigation, we need to load these first:
    <Uri>urn:SIMATIC.S7-1500.OPC-UA.Application:PLC_1</Uri>
    <Uri>http://opcfoundation.org/UA/DI/</Uri>
    <Uri>http://www.siemens.com/simatic-s7-opcua</Uri>

Can you provide these Siemens nodeset XML files?

anxuae, 2022-02-21:
Hi @mikakaraila , I've tested this new feature and it's working great! thank for it.
I've notice one difference compared to a server "manually created": there is not output from server node when a OPCUA variable is changed. What is the reason?

mikakaraila, 2022-02-21:
XML loading will not handle it same way as addVariable.

anxuae, 2022-02-22:
Is there any way to make it working as "addVariable"? It is easier to implement the server logic compared to using client read/write and subscribe node in the server side

mikakaraila, 2022-02-22:
Not today, long day, but later this week I will have some time. I just have to iterate variables and bind function to them.

mikakaraila, 2022-02-24:
Some progress, not yet ready...

mikakaraila, 2022-03-01:
OK, first version almost ready, need to clean up and test more.

mikakaraila, 2022-03-04:
I published new version that will support bindVariables now, please test.

mikakaraila, 2022-03-09:
BindVariable get/set functions only on own node-opcua server, not to another server.
It can be also case that the bindVariable cannot handle all variable.
Can you provide me your nodeset XML file to test this?
As I tested I got similar output msg payloads.

mikakaraila, 2022-03-10:
Anything in the console?

anxuae, 2022-03-10:
hi @mikakaraila 

I've sent you the XML by mail.
There is no errors in the console

anxuae, 2022-03-10:
I've also noticed that it is impossible to write String values in server created with a nodeset:

[image]

anxuae, 2022-03-17:
Hi @mikakaraila 

I noticed a discrepancy between the nodeSet.xml (PLC_1) name space and the namespace defined in the generated server:

[image]

mikakaraila, 2022-03-18:
Nothing to fix because this is normal. 
Namespace index depends on how many other namespace are loaded into the address space.

[image]

anxuae, 2022-03-18:
But I would like to use the same flow for the real PLC (which respect namespace of the XML) and the simulated PLC. If the name space change, All flow shall be updated each time I'm switching between real and simulated PLC.

Where the 4 first namespaces are from? Its seems generated by the node-opcu lib

mikakaraila, 2022-03-18:
Actually correct way to use namespace index is to use urn to find correct namespace index.

In your case one way could be to remove other nodesets so they will not be loaded.

anxuae, 2022-03-18:
I don't understand how can y do it. In the XML, there are only 3 namespaces:

[image]

Why there is more namespace in the created simulator?

mikakaraila, 2022-03-18:
Look with UaExpert what is loaded:
[image]

anxuae, 2022-03-21:
Hi @mikakaraila 

Even with only one nodeset loaded, there are too many adress space present in the server created by loading the XML nodeset file. I belive that the node-opcua lib is responsible for adding those extra namespace.

So if I understood, I have no reliable way to build clients compatible with simulated PLC and real PLC without a fastidious nodes edition. The feature of creating a simulator using nodeset seems compromised for my needs.

mikakaraila, 2022-03-21:
Not with the same namespace index, but Etienne is fixing something.
I have to add order to nodesets that server will load then this will be perhaps possible.

anxuae, 2022-03-21:
Thanks a lots!

And concerning the bind of the variables, the last version of your lib does not solve it (I'm still testing with the XML I've sent you by mail).

mikakaraila, 2022-03-23:
There was too old DI nodeset, I just merged it. Then there was one mistake with String/DateTime conversion.
Please test again.

anxuae, 2022-03-28:
Still have no output when a value change in the server using the v0.2.265

mikakaraila, 2022-03-29:
Etienne had made now some fixes, waiting for new release. But I have to test it also...

mikakaraila, 2022-04-10:
NOTE: Old version saved only server´s own namespace (ns=1). Now msg.topic can be used as namespace index to be saved.

anxuae, 2022-04-11:
Hi @mikakaraila did you have new from Etienne about the bind of the variables?

anxuae, 2022-05-04:
Hi @mikakaraila do you have some news about having output from server node when created using an XML?

mikakaraila, 2022-05-05:
Sorry forgot to test, it could work now as Etienne has updated & fixed something inside node-opcua v2.68.0.
Please re-test.

mikakaraila, 2022-05-06:
Solved, My misunderstanding with server. It cannot load own namespace. This means ns=1 is created as server starts.
So you have to build new variables and folder to other namespace index than 1.
Then this will work and you can load nodeset from the XML file:
[image]

closed 2022-06-29 by mikakaraila

AutomationTeam-ATeam, 2023-12-20:
> Solved, My misunderstanding with server. It cannot load own namespace. This means ns=1 is created as server starts. So you have to build new variables and folder to other namespace index than 1. Then this will work and you can load nodeset from the XML file: [image]

Please can you explain this? Could you also add it to the examples?
I can't load a configuration because the Node-Red server crashes when I try it.
The example of the XML used also be very useful.