mikakaraila/node-red-contrib-opcua #375: Can not use environement variable in URI configuration node

anxuae, 2022-01-20:
Hello,

I would like to use the same subflow for several PLC connection. For this reason, I need to setup different `EndPoint` in different instances of the subflow.
I've tried to use environment variable in the  `EndPoint` of the configuration node without any success (I can do that using with the nodes provided by _**node-red-contrib-modbus**_ library):

[image]

How can I achieve that?

mikakaraila, 2022-01-21:
You can inject new endpoint, but I will check this.

anxuae, 2022-01-21:
I did not find any way to inject a new endoint, the configuration node can not be edited using the `msg `properties

mikakaraila, 2022-01-21:
Look examples folder and OPCUA-TEST-NODES.json

anxuae, 2022-01-21:
Where can I found the `endpoint.json` file which is parsed in the file parser?

anxuae, 2022-01-21:
Ok that you for your answer, I will try it

anxuae, 2022-01-21:
So that means, I define a random URI for all my nodes in subflows, then  I sent a "reconnect" message for each OPCUA nodes with a specific End-Point

anxuae, 2022-01-22:
It seems working, thanks a lot!

closed 2022-01-22 by anxuae