mikakaraila/node-red-contrib-opcua #377: Reset internal node after READ/WRITE ALL

anxuae, 2022-01-22:
Hi @mikakaraila,

I'm still here with my subflow needs. Is it possible to automatically clear internal node items after sending the readAll, writeAll command?

mikakaraila, 2022-01-23:
It is clearitems that you have to use.

[image]

anxuae, 2022-01-23:
Thanks for your answer. How can I chain it with others OPCUA items? I would like to do it automatically before each read/write.

mikakaraila, 2022-01-24:
I don´t understand what you are asking.
First add items into the read multiple, then you can use it to read items as many times as your flow runs.
Clear items is just if you want to make it dynamic and change items.

anxuae, 2022-01-27:
Hi @mikakaraila,

I would like to make reusable subflow to write values OPCUA server nodes in a row.
Here is the sublow that I've thinking about in order to sequentially:
- clear items
- write new values
- trigger write all action

each time a message with new vales is entering in the subflow.

[image]
 

However the actions to `clear items` and `write all` sent to the `Client node` output nothing. Thus I can not chain actions. Can you give me a proper way to do it?

mikakaraila, 2022-01-27:
Typo: clearitem and writemultiple no space in between :-)

Look from the examples OPCUA-TEST-NODES.json!

anxuae, 2022-01-27:
Sorry for mistake, it is just in the issue. Event without space. there is nothing at the output of the client node when action `clearitems `and `writemultiple `. So I can no chain the nodes

mikakaraila, 2022-01-27:
Flow works only when server (flow) is started:
[image]

You have to enable some endpoints like None etc. to access server:
[image]

anxuae, 2022-01-27:
There is no OPCUA server in the attached flow file. Did you download the correct file?

mikakaraila, 2022-01-27:
Sorry, now I have your flow:
[image]

mikakaraila, 2022-01-27:
Here quick fix that will show you how to add variables and then write values:
[image]

I just added simple function node to write payload that contains array of nodeIds and values.
[image]

[flows.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7952802/flows.txt)

anxuae, 2022-02-03:
Hi @mikakaraila 

I've got an error using you inject node:

[image]

If I remove the topic "exemple", nothing is written

mikakaraila, 2022-02-03:
Change topic to "ns=1;s=example" I expect code will check it is valid nodeId.

anxuae, 2022-02-04:
Hi @mikakaraila thanks, it's working perfectly! And there is no need to set the topic with "writemultiple".

It seems that this is not possible with "READ Multiple": I sent an array of node to read. and msg.topic="ns=1;s=example", only the node defined in the topic is read but not the one in the `msg.payload` array

mikakaraila, 2022-02-16:
OK => closed.

closed 2022-02-16 by mikakaraila