mikakaraila/node-red-contrib-opcua #378: OPC UA Certificate Authentication

keremsubasi, 2022-01-24:
Hello,

I'm having a problem while trying to connect an OPC UA Server to the Client on Node-Red.
I'm using an Siemens IOT2050 with Debian.

The Server is provided by an Heidenhain PLC and only allows authentication over certificates an private keys.

Security & Authentication Settings:
Security Mode: Sign&Encrypt
Security Policy:  Basic256Sha256
Authentication: X.509 Certificates

For that I created the certificate (.der) and the private key (.pem) with openSSL.

The OPC UA Server I'm using provides three more file which need to be stored in the OPC UA Client:
A server certificate (.der), a certificate revocation list (.crl) and a rootca certificate (.der).

I was already able to get a connection with UaExpert, but I´m not sure, where to store all the certificates in node-red.
In UaExpert i used the user certificate and the private key (which I created) to authenticate and also imported the certificate into the server.
I stored the Server certificate in ..\uaexpert\PKI\trusted\certs, the rootca in ..\uaexpert\PKI\issuers\certs and the .crl file in ..\uaexpert\PKI\issuers\crl.
I also created an client application certificate in UaExpert and imported it into the OPC UA Server.

Is this possible with the node red node? 

Do I have to create a client certificate and where do I store it?
Where do I need to store all the files I get from the OPC UA Server?

I can`t find a folder with a name like "node-opcua-client" or "PKI".

I couldn't find any instruction for that and hope that someone can help me with my problem.

Thanks in advance,
Kerem

OriolFM, 2022-01-24:
I'm having some connection problems of my own, but I think I can help there.

If you are running it in a Linux server, the PKI folders will probably be in /root/.config/node-red-opcua-nodejs and /root/.config/node-opcua-default-nodejs.

Inside PKI, there's a folder ./own/certs , and I guess that's where your certificate goes. I'm using self-signed certificate and that's where it is, anyway.

mikakaraila, 2022-01-25:
You cannot generate certificate on the other computer as it contains hostname of the computer.
Certificate will not match.

This is my Windows environment default certificates:
[image]

keremsubasi, 2022-01-25:
I don't have this UserPKI folder. Can I just create it or won't that work?

mikakaraila, 2022-01-25:
If you use server node it will create it and use it.

keremsubasi, 2022-01-25:
Okay thanks, I have a UserPKI folder now. But the folder is in /root/.config/node-red-opcua-nodejs/ and not /root/.config/node-opcua-default-nodejs/. Does that matter? 

Do I store the user certificate in the own folder?

mikakaraila, 2022-01-25:
Normally node-opcua certificate manager takes care of everything...

You should use it and then if you want to put your own certificate just replace files it has created.

keremsubasi, 2022-01-28:
I couldn't find any user certificate in the UserPKI folder that I could replace. So I created a user certificate with the following commands:

```
openssl genrsa -out ./user_nd_key.key 2048
openssl rsa -in ./user_nd_key.key -text >user_nd_key.pem
openssl req -x509 -sha256 -days 3650 -key ./user_nd_key.key -outform DER -out user_nd.der
openssl x509 -inform der -in user_nd.der -out user_nd.pem
```

So I got the the certificate as .pem and .der and the private key as .pem. I stored the user certificate and key in /root/.config/node-red-opcua-nodejs/UserPKI/ in the certs and private folder. I exported the .der file to the server.

I converted the server certs from der to pem and stored them in the PKI folder in trusted and issuers.

I also created a client certificate and put the .pem cert in cert folder and the .pem key in private folder: 

```
openssl genrsa -out ./client_nd_key.key 2048
openssl rsa -in ./client_nd_key.key -text >client_nd_key.pem   
openssl req -x509 -sha256 -days 3650 -key ./client_nd_key.key -outform DER -out client_nd.der
openssl x509 -inform der -in client_nd.der -out client_nd.pem
```
I exported the .der file to the server.

I configured the client node like this:

[image]

Is that correct?

My Problem now is, when I try to connect  to the server, I get "invalid endpoint Error: server Certificate verification failed with err BadSecurityChecksFailed (0x80130000)"

mikakaraila, 2022-01-28:
Console output would help me to provide some instructions...

keremsubasi, 2022-01-28:
What do you need exactly?

mikakaraila, 2022-01-28:
This from your node-red console:
[image]

mikakaraila, 2022-01-28:
You have already server running on same port!

**Error port in use**

keremsubasi, 2022-01-28:
How do I disable/deactivate that?

mikakaraila, 2022-01-28:
Do you have 2 node-red instances running at the same time? Close the other one or take console from the running one...

keremsubasi, 2022-01-28:
I don't think that I have 2 instances running. Honestly,  I don't know what to do.

mikakaraila, 2022-01-28:
Can you use Linux command line commands?

ps -ef

I expect you have node-red running as service

keremsubasi, 2022-01-28:
[image]

[image]

mikakaraila, 2022-01-28:
You have node-red already running:
[image]

keremsubasi, 2022-02-07:
Hello,

I still have problems with setting up the client. Can you please tell me, which format I have to use for the certificates and the keys or doesn't that matter?

I have this files:

**Server:**
certificate.der
rootca.der
CertificateRevocationList.crl

I would like to create my own client cert and key and the user cert and key. Which formats should I use for this?

Last question is where to store these files? There are two PKI folders. One is in /root/.config/node-red-opcua-nodejs and the other one /root/.config/node-opcua-default-nodejs.
Which one is correct?

Thanks

mikakaraila, 2022-02-07:
I have Windows environment only at the moment.

It places certificates under node-red-opcua-nodejs folder in pem format.
I expect node-opcua-pki can handle der & pem formats.

This is just example Iám using PKI Storage folder: MYSERVER.
[image]

keremsubasi, 2022-02-08:
Does that mean, that I don't use the same certificate to import in the server which I use for the client? And do I have to delete the certificate for the server from this folder afterwards?

keremsubasi, 2022-02-09:
I'm currently testing it on my windows environment and I think I am getting closer to a successful connection.

I stored all the certificates in the PKI folder and configured the Client node like this:

[image]

But now I get this error in the command prompt:

[image]

Does that mean, that I need the subjectAltName "applicationUri  = urn:Kerem:NodeOPCUA-Client" in my User Certificate? Do you know how I can create this?

mikakaraila, 2022-02-09:
Look config file that defines all parameters for the certificate:
https://github.com/node-opcua/node-opcua-pki/blob/master/bin/crypto_create_CA_config.example.js

It is not so simple to generate certificate with all needed features.

mikakaraila, 2022-02-10:
As I tried to tell you it will require extra work to get additional information that you want to add into the certificate.
The current implementation provided by node-opcua-pki is general solution.
If you need help then I advise you to contact @erossignon for commercial support.

keremsubasi, 2022-02-15:
Hello @mikakaraila, sorry to disturb you again, but I still cant connect the server. Is it maybe possible that we arrange a meeting and you give us a little support. 

I am sure that the problem is not the user certificate. The problem could be, that we are not able to communicate with a hostname. So I created a client and user cert with a URI like this: URI:urn:192.168.204.xxx:NodeOPCUA-Client. I stored the client cert in 
[image]

and the user cert in 
[image]

NodeRed configuration is like this: 
[image]

Unfortunately I couldn't reach to @erossignon so I would really appreciate your help, since I have this problem already for a month. 
Thanks in advance 
Kerem

mikakaraila, 2022-02-15:
Missing subjectAltName, if you use command line then you should provide correct config.js that will contain it.
You cannot generate certificate with all needed parameters on command line.

Code snippet that will generate error message:
```Javascript
 if (!uniformResourceIdentifier) {
        warningLog(
            `[NODE-OPCUA-W14] The certificate subjectAltName uniformResourceIdentifier is missing.\n` +
                `Please regenerate a specific certificate with a uniformResourceIdentifier that matches your ${type} applicationUri\n` +
                `applicationUri  = ${applicationUri}\n` +
                `certificateFile = ${certificateFile}\n`
        );
        return;
```

keremsubasi, 2022-02-15:
I used a config file with this content:

```
[ req ]
default_bits = 2048
default_md = sha256
distinguished_name = subject
req_extensions = req_ext
x509_extensions = req_ext
string_mask = utf8only
prompt = no

[ req_ext ]
basicConstraints = CA:FALSE
#nsCertType = client, server
keyUsage = nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign
extendedKeyUsage= clientAuth, serverAuth
nsComment = Kerems Client Cert"
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid,issuer
subjectAltName = URI:urn:192.168.204.180:NodeOPCUA-Client,IP: 192.168.204.180

[ subject ]
countryName = DE
#stateOrProvinceName = Hessen
localityName = Gladenbach
organizationName = ACS Deutschland GmbH
commonName = Kerem
```

and the commands for the client cert I used are like this:
```
openssl req -x509 -days 365 -new -out keremclient.pem -keyout keremclient_key.pem -config client.conf
openssl rsa -in keremclient_key.pem -out keremclient_key.pem
openssl x509 -inform PEM -outform DER -in keremclient.pem -out keremclient.der
```
Is that wrong?

One thing I noticed, is that when I try to connect with my own certificate, a new one is generated in the trusted folder:
[image]

mikakaraila, 2022-02-15:
Please attach trusted pem file here I will take a look into the content.

And message in the verify is actually **warning** so certificate should work, it is just reporting missing subjectAltName.

keremsubasi, 2022-02-15:
[client_and_user_cert.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/8069522/client_and_user_cert.zip)

In this folder you will find the user and client certificates I created.

keremsubasi, 2022-02-16:
Hey, were you able to check the certificates?

I started node red on the command prompt with the command: node-red -v
Maybe this helps you better to find the problem. This is what I got on the command prompt:

[image]

In the last rows it says " Case B: UserName & password does not match to server (needed by Sign): undefined undefined". But I don't connect with username and password but with user certificate.

keremsubasi, 2022-02-16:
Is it even possible to use another user authentication methode than anonymous or  username&password? 
I thought it is, because it is mentioned in the node-opcua readme (https://github.com/node-opcua/node-opcua#readme):
[image]

If that is not possible, I can't connect to my Server anyway.

mikakaraila, 2022-02-16:
You have misunderstood options, you can use:
1) Anonymous (None/None), no username/password and no encryption.
2) Sign without Encrypt => This means username & password.
3) Sign & Encrypt => username & password with encrypt (certificates)

You can check you server endpoints easily with UaExpert this way:
[image]

I will check you certificate next.

keremsubasi, 2022-02-17:
Yes that is the security mode, but I am talking about the authentication method. My server only allows authentication with certificates an private key.  And I thought that this two fields are for the user cert and key:

[image]

This is for example how I configure it in UaExpert. And I thought that this is possible with the client node aswell.
[image]

If I am correct, this authentication method needs to be implemented first and than I could connect to my server.

mikakaraila, 2022-02-17:
With certificate you must always provide username and password when you open session.
Check this: https://opcfoundation.org/wp-content/uploads/2014/08/11_OPC_UA_Security_How_It_Works.pdf page 25.

But I can take deeper look next week. It can be something as normally everything is done with certificate manager.

keremsubasi, 2022-02-18:
I think this only applies on authentication with username and password. Authentication with Certificates should be independent to that. 

There are 3 ways to authenticate (at least with UaExpert):
-Anonymous
-Username & password
-Certificate & private key

And if you choose certificate & private key, you shouldn't need an additional username & password. In my Server there isn't even an option to set an username & password because it only provides authentication with certificates.

With UaExpert you don't need an additional username & password, if you login/authenticate with certificate & private key. 

And in the OPC UA client node there is no option to store a user authentication certificate and private key (like there is in my UaExpert Screenshot).

mikakaraila, 2022-02-18:
Ok, I think now it is not implemented / supported at least not by node-red-contrib-opcua.
I have to check if it is possible by node-opcua.

keremsubasi, 2022-02-18:
Yes, that is what I meant with this comment: https://github.com/mikakaraila/node-red-contrib-opcua/issues/378#issuecomment-1041384102

I think it is available/supported in node-opcua and needs to be implemented into node-red-contrib-opcua.

Thank you very much for your help Mika!

keremsubasi, 2022-02-18:
Ok, thanks a lot.

keremsubasi, 2022-03-03:
Hello Mika,

did you have to opportunity to check, how the new authentication method could be realized? 

Best regards
Kerem

mikakaraila, 2022-03-03:
Not yet, I just finished bind variables and bind method for the server.
I will look this later today or tomorrow.

mikakaraila, 2022-03-06:
One problem, to test this I need a server that uses only certificate for the authentication.
Are there any available if not I have to implement one for the testing. Prosys doesn´t allow certificate alone authentication.
I have to take a look to specification also.

keremsubasi, 2022-03-06:
Sorry, can’t help you with that. I want to use it with my PLC so I don’t know about any Test server. 
But isn’t it possible to select “certificate” for user authentication methods at the users tab in Prosys? I haven’t tested that yet.

mikakaraila, 2022-03-06:
No as it requires: select at least one

[image]

keremsubasi, 2022-03-06:
Isn’t that selected here?
[image]

mikakaraila, 2022-03-06:
It is **IssuedToken** not a certificate!

Node-opcua has some support for it, I just have to study it a little bit...

keremsubasi, 2022-03-06:
No that is just a screenshot I took from the manual because I don’t have my pc with me. I need authentication with certificate and private key. Not the issuedtoken

keremsubasi, 2022-03-06:
So user authentication with certificate like this 
[image]

keremsubasi, 2022-03-06:
Connecting to a server which provides authentication with certificate and password looks like this in UaExpert:
[image]

that’s a screenshot I posted earlier in this issue thread.

mikakaraila, 2022-03-07:
That is normal certificate based authorization, but do you know what you are asking?

If we want to use Issuer token then you should provide authorization service, do you have it?
Read this:
https://reference.opcfoundation.org/v104/Core/docs/Part4/7.37/

keremsubasi, 2022-03-07:
Yes I need the normal certificate based authentication. Nothing with Issuer token.
So instead of connecting anonymously or with a username and password, I need to authenticate with certificate and private key.

I'm sorry that I confused you with the screenshot but that is just from the prosys manual. (I wanted to show where you could select the certificate based authentication.)

[image]
These are the security settings of my opc ua server. So just normal certificate and private key authentication. No issuer token.

mikakaraila, 2022-03-07:
It is in your reply Security Mode: **Sign&Encrypt** it is normal certificate based authentication.
You have to use username & password with the certificate.

keremsubasi, 2022-03-07:
No it says Authentication: X.509 Certificate.

I can't set a username and password on my server. I don't have that option. 

Sign&Encrypt is the Security mode not the authentication setting.

Please check this screenshot from UaExpert.
There is the option to store a **user certificate and private key**. Your Client doesn't have that and that is what I am asking for. I can't connect with a username&password, because I have no username and password.
[image]

And this is how your client node has to look like, so I can connect my server:
[image]

keremsubasi, 2022-03-07:
Please check this out: https://reference.opcfoundation.org/Core/docs/Part4/7.41.1/

There it says that OPC UA currently supports four different ways to authenticate a user.

Your client currently provides 2 of the 4 ways, anonymously and with username & password. And what I need is the third option: 
`X509IdentityToken | A user identified by an X.509 v3 Certificate.`

mikakaraila, 2022-03-07:
OK I expect you want to get this working. It will need change in UI and code.
```Javascript
const userIdentity: UserIdentityInfoX509 = {
            certificateData: clientCertificate,
            privateKey,
            type: UserTokenType.Certificate,
        };
```

Not this week, I have so much work to do at the office.

keremsubasi, 2022-03-07:
Yes, that should be it! Thanks Mika!

mikakaraila, 2022-03-07:
Got some time as teams meeting following is easy:
[image]

keremsubasi, 2022-03-07:
Yes, that is exactly what I meant! Perfect, thanks.

mikakaraila, 2022-03-07:
Still testing, something missing from the user certificate, signature is invalid.
Trying to test this before publishing new version.

mikakaraila, 2022-03-08:
I have to add some instructions how to generate user certificate. I tested how to use it with UaExpert and node-red server node.
Node-red client node needs still some modifications.

keremsubasi, 2022-03-28:
Hello Mika,

Thank you very much for the new implementation. 
I'm currently testing it and I have a problem with the connection from my OPC UA Server to the Node-Red Client Node. 
In the command prompt it says "verify that server certificate is trusted or that server certificate issuer's certificate is present in the issuer folder" but I stored the server certificate into the trusted folder and the root certificate to the issuer folder.

Do I have to store the user certificate and private key in a specific folder or can I store store it anywhere as long as I enter the correct path?

Here is a screenshot from the command prompt:
[image]

mikakaraila, 2022-03-28:
Now you have to use UserPKI as it is own specific certificate instead of normal certificate. My computer Windows uses this one:
C:\Users\karaimi\AppData\Roaming\node-red-opcua-nodejs\Config\UserPKI\trusted\certs

[11 more replies]

closed 2022-04-04 by keremsubasi