mikakaraila/node-red-contrib-opcua #380: new version problem: cannot read property 'position' of undefined

Eldarion85, 2022-01-27:
Hello,

I have been using this opcua client for 3 years on three different raspberry-pi with different systems (Stretch, Buster, Bullseye). It's great that the client is always being developed further and so far I've always been able to help myself with every problem or it was fixed quickly.

Unfortunately, I've now encountered a problem that I can't solve here, despite the issues (https://github.com/mikakaraila/node-red-contrib-opcua/issues/289) being closed.

My error is:
invalid endpoint TypeError: Cannot read property 'position' of undefined

I use my own root CA and used it to create a client certificate for node-red and a server certificate for an S7-1500. I have to do professionally with the creation of x509 certificates and know my way around very well. I am also very familiar with the S7-1500 and look after a lot of opcua connections.

Of course, I checked all points from the issues above again. My certificates are valid (openssl x509 -in "certs.pem" -text) and the rejected/certs folder is empty! The strange thing is that i have been using exactly the same certificates for 2 years and they work perfectly on the other systems with older versions. Each client has its own certificate, of course, but the structure is exactly the same.

My root ca is in the folder:
/home/iobroker/.config/node-red-opcua-nodejs/PKI/issuers/certs

My crl from this root ca is in the folder:
/home/iobroker/.config/node-red-opcua-nodejs/PKI/issuers/crl
(of course the crl is still valid)

My public-key is in the folder:
/home/iobroker/certs/public

My private-key is in the folder:
/home/iobroker/certs/private

Since the new update you can also specify the path to the PKI folder in node red. I think that's very good, but for testing I left it the same way it runs on all other systems without problems.
/home/iobroker/.config/node-red-opcua-nodejs

In the s7-1500 I added and loaded my root-ca as trustworthy. Exactly as it also works with the other connections.

For testing, i switched off the new pi and placed the certificates on an old one in the same directories and connected them to the same s7-1500. The connection is established immediately. Something must have changed, which is why the connection is no longer established using the same procedure and the same valid certificates.

Many thanks in advance for the support!

mikakaraila, 2022-01-27:
Openssl has been updated and node-opcua-pki

Can you attach a bit more information that I could see what causes position is undefined error message?

mikakaraila, 2022-01-28:
Another cause could be too old nodejs. Node-opcua requires new nodejs if you are using still some "really" old one.

Eldarion85, 2022-01-28:
I had already tried that with the NPM cache. I would like to give more information, but where can I find it? Only the message from the topic is displayed under the node and nothing is unusual in the debug.

After your message yesterday, I first backed up the Pi and then restored it from a backup before Node-Red was installed. After that, I put my scripts back into the appropriate folders as described above. The connection was established immediately and everything works perfectly.

I can't say where the problem was right now, but I'll check again as soon as I have time. I only did one thing differently after restoring from backup. I don't know how familiar you are with the node-red in ioBroker. It is possible to install new nodes via the ioBroker adapter or via the palette manager directly in Node-Red. On the first installation I installed the OPC client via ioBroker, on the second I activated the palette manager and did it directly from Node-Red. I've used both options in the past and haven't had any problems, so I can't say if that was the reason now. For me, the problem is solved for now, but I'll write when I've found out what actually caused it.

Thank you!

Eldarion85, 2022-01-28:
I completely reinstalled the Pi.

npm -v
6.14.15

nodejs -v
v14.18.3

node -v
v14.18.3

Also, I restored the Pi from backup after iobroker, npm, nodejs everything was already installed. I only installed node-red and the opc client and put the certificates back in the folders.

mikakaraila, 2022-01-28:
Looks good, have you tried to first clean npm cache and then re-install node-red-contrib-opcua?

Eldarion85, 2022-01-28:
Yes, I did it exactly as described in your screenshot above and in the other issue. That didn't help. After the re-install I had the same message again.
Even if it works now, I would of course like to know what caused it. my only approach is what I did differently on the second try. I will restore the pi twice more from the same backup. Once i install the opc client in node-red again via the iobroker and once via the palette manager in node-red. If that leads to the fact that it only runs via the palette manager, that is at least a starting point for troubleshooting.

mikakaraila, 2022-01-28:
I still expect that there is some package mismatch. 
Command "npm dedupe" could also be used to solve it.

mikakaraila, 2022-02-02:
At my Windows I use "own" PKI store folder "MYSERVER" and here is server certificate:
[image]

Normal config use PKI store "Config":
[image]

I expect some changes on node-opcua have changed "default" folder name camel case as I use default certificate manager:
[image]

Eldarion85, 2022-02-02:
I have now also tested it with my own PKI folder. That works fine. I haven't had any problems since I reinstalled.
I only store the CAs of the servers in the PKI folder (PKI/issuers/certs). I don't use the "PKI/own" folder because I store the client certificate and private key directly in my own user folder (home/iobroker/certs) and specify the path in node-red.

[image]

In my node-red installation under iobroker i only find one folder "node-red-opcua-nodejs". But I think the error I had is not the wrong folder. My client certificates are now in exactly the same folders as in the first installation where i had the problem. The ca that issued the server certificate for the s7-1500 is also in the same folder.

[image]

[image]

mikakaraila, 2022-02-02:
Certificate file names are not matching:
[image]

[image]

Eldarion85, 2022-02-02:
I have accessed the folder structure below for information only.

The certificate
/home/iobroker/.config/node-red-opcua-nodejs/PKI/own/certs/client_certificate.pem

and the private key
/home/iobroker/.config/node-red-opcua-nodejs/PKI/own/private/private_key.pem

are only used if I select "None, use generated self-signed certificate" in node-red!

[image]

I use a separate folder for the client certificate and specify the path in node-red. I select "Local certificate file, give path with filename (pem)". Of course this path and the name of the certificate are correct. With this option, the certificates from the "PKI/own" folder are not used.

[image]

I also see on the server that the opc-client registers there with my certificates from my own folder. If I choose the other option, the certificate will be sent to the server, which is located in the "PKI/own" folder.

mikakaraila, 2022-02-02:
Back to original problem:  invalid endpoint TypeError: Cannot read property 'position' of undefined

Install this npm package and use command line with option dump to check/validate your certificate.
https://www.npmjs.com/package/node-opcua-pki

node-opcua uses this inside certificate manager and it uses openssl to dump certificate.

Eldarion85, 2022-02-02:
This is the certificate i use on my test client. With the certificate the connection to my test-s7-1500 has been running for 3 years without problems. I just extended the validity every year and reissued it.

[image]
[image]

Eldarion85, 2022-02-02:
I wrote above that I only installed the opc client in two different ways:

> I can't say where the problem was right now, but I'll check again as soon as I have time. I only did one thing differently after restoring from backup. I don't know how familiar you are with the [...]

I have now reinstalled node-red and the opc client again and installed the npm module via the iobroker. So how it came to the error during the first installation. Unfortunately, I can no longer reproduce the problem, it worked immediately this time too.

mikakaraila, 2022-02-02:
Either it was npm cache or other npm package manager problem.

Eldarion85, 2022-02-02:
> Openssl has been updated and node-opcua-pki
> 
> Can you attach a bit more information that I could see what causes position is undefined error message?
> 
> Earlier case was somehow related to npm cache, after this it started to work: [image]

I had previously tried all the points in the same way and that didn't help me.
But after that it now works with two new installations without problems, so we can probably close this here for now.
If I encounter the problem again, I will investigate it more closely and get back to you if necessary.
Thank you very much!

closed 2022-02-02 by Eldarion85