mikakaraila/node-red-contrib-opcua #381: Question: Opc ua Browse endpoint URL via inject node

henjoe, 2022-01-28:
Hi Mika!

It's been a long time! 
Just want to know if it is possible to configure the property of OPCUA browser via an inject node? 
Currently we set the endpoint manually, but can we do it dynamically during runtime?

[image]

Like for example in the above image, the endpoint URL was configure manually, 
I hope we could set the endpoint URL thru other node or inject node via for example (msg.endpointURL = "someURL")

Thank you!

Regards,
Henjoe

mikakaraila, 2022-01-28:
Look examples and OPCUA-TEST-NODES.json Reconnect tab.

[image]

[image]

henjoe, 2022-01-28:
There you go!

Thanks Mika!

Regards,
Henjoe

closed 2022-01-28 by mikakaraila

henjoe, 2022-01-30:
> <mikakaraila/node-red-contrib-opcua/issue/381/issue_event/5968510201@
> github.com>

Hi Mika,

I tried your instruction but my debug node has this error:
[image: image.png]

here is my function:
[image: image.png]

Hoping for your guidance.

Regards,
Henjoe

henjoe, 2022-01-31:
Additional information: 

This is my function node for browsing: 
[image]

This is my opcua-browser config: 
[image]

My flow: 

[image]

The debug / Result once inject node was pressed.
[image]

henjoe, 2022-01-31:
Hi Mika,

I figure out the error.
My firewall is blocking the nodes and the opcua server that's why there is node data comming out.

Regards,
Henjoe