mikakaraila/node-red-contrib-opcua #383: ExtensionObject in method calls

yasdfgr, 2022-02-01:
I want to call a method with an input argument which is a ExtensionObject (data structure here: https://reference.opcfoundation.org/v104/AutoID/v101/DataTypes/ScanSettings/)

I tried serveral things and it seems that i am on the right way - type id is found on the opc server (in contrast if i test other values)

This is the input argument for the method block:
```
scanSettingsParams = {
            duration: 0,
            cycles : 0,
            dataAvailable : false,
            locationType: 0
};

msg.inputArguments=[{name:"ScanSettings", dataType: "ExtensionObject", typeid:"ns=1;i=3010",
value:JSON.stringify(scanSettingsParams)}];
```
But always i get the error from the method block: "Invalid NodeId: TypeError: Cannot read property 'name' of null"

In the source file i think in line 254 the variable name is wrong "extensionobject" vs. "extensionObject".
But this is not solving the problem, i debugged it - the node.session.constructExtensionObject() function fails everytime with throwing the error above.

The parameters are definitly right. I also found this SO post: https://stackoverflow.com/questions/61266294/call-opcua-method-with-struct-input-argument-using-node-opcua which is exactly describing what i want to do. And the code is very similar.

https://github.com/mikakaraila/node-red-contrib-opcua/blob/1a837a09cd938b0be6c6de2c8f842ec9807251e6/opcua/107-opcuamethod.js#L251-L261

mikakaraila, 2022-02-02:
Did you notice that you have to have AutoID types loaded and namespace index can be something else...

[image]

mikakaraila, 2022-02-02:
Here is example of method call:
[image]

yasdfgr, 2022-02-02:
> Did you notice that you have to have AutoID types loaded and namespace index can be something else...
> 
> [image]

yasdfgr, 2022-02-02:
"Did you notice that you have to have AutoID types loaded and namespace index can be something else..."

Hmm, "load AutoID types" - not clear for me. Where can i find the example from your screenshot?
My method call itself looks like yours.

mikakaraila, 2022-02-02:
Zoom:
[image]

yasdfgr, 2022-02-02:
Do you have the node-red flow code which is show in your screenshot. I don't know what to do, i need a example. Thank you.

mikakaraila, 2022-02-02:
Here:

[CUSTOM_OBJECT.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/7985788/CUSTOM_OBJECT.txt)

yasdfgr, 2022-02-02:
Regardless what i do - it doesn't work.
I did a wireshark analysis, because with uaExpert everything works.
[image]
I see that the method call sends the inputargument with ns3;i=5015.
I found also some information in the sources of a working opc ua client about available nodeId's what has to do with ScanSettings.

```
/// The identifier for the ScanSettings DataType.
ScanSettings = 3010;

/// The identifier for the Default Binary Object.
ScanSettings_Encoding_DefaultBinary = 5015;

/// The identifier for the Default XML Object.
ScanSettings_Encoding_DefaultXml = 5016;

/// The identifier for the ScanSettings Variable.
AutoID_BinarySchema_ScanSettings = 6044;
```

Can you explain how this things are linked together? Im confused...
Until now i think that the type of extensionObject is i=3010. I expect that the extensionObject with this type is build in the background with the value (=parameters in json format) by the client and used together with the method call.
[image]

I don't understand what the BUILD command i your example should do. It writes some data to the type object?

How can i force that the argument will be correctly build and sent as ns=3;i=5015 by the method call, as i can see in the wireshark?

mikakaraila, 2022-02-02:
I have Siemens RF360R just now no time to play with it.

BUILD is used at the server side to get actual type structure. Then it could be used on client to set values inside the method.
This is not yet final/complete implementation, I have work to do at the office, bit busy time again...

federicab20, 2022-02-17:
Hello
I'm studying recently the OPC-UA protocol and the library node-red-contrib-opcua, so I don't understand CUSTOM_OBJECT.txt  example. The idea is to create a new structure that contains the values ​​of the ScanSetting structure present in DataTypes folder as the following image. Is the first part correct?

[image]

In the first flow of Node-Red, in Server Custom Nodesets node, you have written the field 'custom nodeset directory', but which file is present in this path? The XML file?
[image]

In the second part of the flow, you have set the 'Build' action for the Client, why is this action necessary if the 'Test' variable is created with the first command?

Another question: why the 'value' field doesn't contain the ScanSetting fields (duration,cycles,dataAvailable)?
[image]

[image]

If I want to create several structure instances, what do I do?

Thank you so much

mikakaraila, 2022-02-17:
Example that is inside OPCUA-TEST-NODES works, but it is perhaps not enough.
I will do separate flow with comments to explain it more. I will have time next week to do it.

federicab20, 2022-02-17:
Yes, I used that example as basis for implementing my code, but I don't understand very well what I have to do. 

The purpose is to create a structure as shown in the image below, and set variable1 = baseObjectType_Xml and variable2 = baseObjectType_Xml, I used the following flow but I get the error.

[image]

[image]