mikakaraila/node-red-contrib-opcua #384: homeassistant - nodered - "invalid endpoint Error"

teun2021, 2022-02-06:
Hi @mikakaraila ,

Fresh install from homeassistant OS on VM gives me an error.
"Invalid endpoint error: Command failed: "openssl" rand -out "............  openssl: not found"

Yet, on the same machine opcUA on Docker works.
I did 4 fresh installs of home assistant each with some different settings, non worked.

My 5th install is below, now with virtualbox and homeassistant.
Any idea?

[image]

_My Logfile;_

> Welcome to Node-RED
> ===================
> 7 Feb 00:06:56 - [info] Node-RED version: v2.2.0
> 7 Feb 00:06:56 - [info] Node.js  version: v16.13.2
> 7 Feb 00:06:56 - [info] Linux 5.10.91 x64 LE
> 7 Feb 00:06:57 - [info] Loading palette nodes
> 2022/02/07 00:06:57 [error] 542#542: *161 connect() failed (111: Connection refused) while connecting to upstream, client: 172.30.32.2, server: a0d7b954-nodered, request: "GET /comms HTTP/1.1", upstream: "http://127.0.0.1:46836/comms", host: "homeassistant.local:8123"
> 2022/02/07 00:06:58 [error] 542#542: *161 connect() failed (111: Connection refused) while connecting to upstream, client: 172.30.32.2, server: a0d7b954-nodered, request: "GET /comms HTTP/1.1", upstream: "http://127.0.0.1:46836/comms", host: "homeassistant.local:8123"
> 7 Feb 00:06:58 - [info] Dashboard version 3.1.5 started at /endpoint/ui
> BAIL ON node-red-node-pi-gpio/rpi-gpio
> 7 Feb 00:06:58 - [info] Settings file  : /etc/node-red/config.js
> 7 Feb 00:06:58 - [info] Context store  : 'default' [module=memory]
> 7 Feb 00:06:58 - [info] User directory : /config/node-red/
> 7 Feb 00:06:58 - [warn] Projects disabled : editorTheme.projects.enabled=false
> 7 Feb 00:06:58 - [info] Flows file     : /config/node-red/flows.json
> 7 Feb 00:06:58 - [info] Server now running at http://127.0.0.1:46836/
> 7 Feb 00:06:58 - [info] Starting flows
> 7 Feb 00:07:02 - [info] Stopping flows
> 7 Feb 00:07:02 - [info] Stopped flows
> 7 Feb 00:07:02 - [info] Starting flows
> 7 Feb 00:07:02 - [red] Uncaught Exception:
> 7 Feb 00:07:02 - [error] TypeError: RED.settings.functionGlobalContext.get is not a function
>     at map (/opt/node_modules/node-red-contrib-actionflows/actionflows/actionflows.js:270:51)
>     at EventEmitter.runtimeMap (/opt/node_modules/node-red-contrib-actionflows/actionflows/actionflows.js:593:5)
>     at EventEmitter.emit (node:events:390:28)
>     at start (/opt/node_modules/@node-red/runtime/lib/flows/index.js:383:12)
> [00:07:02] INFO: Starting Node-RED...
> > start
> > node $NODE_OPTIONS node_modules/node-red/red.js "--settings" "/etc/node-red/config.js"
> 7 Feb 00:07:02 - [info] 
> Welcome to Node-RED
> ===================
> 7 Feb 00:07:02 - [info] Node-RED version: v2.2.0
> 7 Feb 00:07:02 - [info] Node.js  version: v16.13.2
> 7 Feb 00:07:02 - [info] Linux 5.10.91 x64 LE
> 7 Feb 00:07:02 - [info] Loading palette nodes
> 2022/02/07 00:07:03 [error] 542#542: *167 connect() failed (111: Connection refused) while connecting to upstream, client: 172.30.32.2, server: a0d7b954-nodered, request: "GET /comms HTTP/1.1", upstream: "http://127.0.0.1:46836/comms", host: "homeassistant.local:8123"
> 2022/02/07 00:07:04 [error] 542#542: *167 connect() failed (111: Connection refused) while connecting to upstream, client: 172.30.32.2, server: a0d7b954-nodered, request: "GET /comms HTTP/1.1", upstream: "http://127.0.0.1:46836/comms", host: "homeassistant.local:8123"
> 7 Feb 00:07:04 - [info] Dashboard version 3.1.5 started at /endpoint/ui
> BAIL ON node-red-node-pi-gpio/rpi-gpio
> 7 Feb 00:07:04 - [info] Settings file  : /etc/node-red/config.js
> 7 Feb 00:07:04 - [info] Context store  : 'default' [module=memory]
> 7 Feb 00:07:04 - [info] User directory : /config/node-red/
> 7 Feb 00:07:04 - [warn] Projects disabled : editorTheme.projects.enabled=false
> 7 Feb 00:07:04 - [info] Flows file     : /config/node-red/flows.json
> 7 Feb 00:07:04 - [info] Server now running at http://127.0.0.1:46836/
> 7 Feb 00:07:04 - [info] Starting flows
> 7 Feb 00:07:36 - [info] Stopping flows
> 7 Feb 00:07:36 - [info] Stopped flows
> 7 Feb 00:07:36 - [info] Starting flows
> 7 Feb 00:07:36 - [info] [OpcUa-Client:53de2f04d7b3701d] 	Local 'own' certificate is NOT used.
> 7 Feb 00:07:36 - [red] Uncaught Exception:
> 7 Feb 00:07:36 - [error] TypeError: RED.settings.functionGlobalContext.get is not a function
>     at map (/opt/node_modules/node-red-contrib-actionflows/actionflows/actionflows.js:270:51)
>     at EventEmitter.runtimeMap (/opt/node_modules/node-red-contrib-actionflows/actionflows/actionflows.js:593:5)
>     at EventEmitter.emit (node:events:390:28)
>     at start (/opt/node_modules/@node-red/runtime/lib/flows/index.js:383:12)
> [00:07:36] INFO: Starting Node-RED...
> > start
> > node $NODE_OPTIONS node_modules/node-red/red.js "--settings" "/etc/node-red/config.js"
> 7 Feb 00:07:36 - [info] 
> Welcome to Node-RED
> ===================
> 7 Feb 00:07:36 - [info] Node-RED version: v2.2.0
> 7 Feb 00:07:36 - [info] Node.js  version: v16.13.2
> 7 Feb 00:07:36 - [info] Linux 5.10.91 x64 LE
> 7 Feb 00:07:37 - [info] Loading palette nodes
> 2022/02/07 00:07:37 [error] 542#542: *172 connect() failed (111: Connection refused) while connecting to upstream, client: 172.30.32.2, server: a0d7b954-nodered, request: "GET /comms HTTP/1.1", upstream: "http://127.0.0.1:46836/comms", host: "homeassistant.local:8123"
> 2022/02/07 00:07:38 [error] 542#542: *172 connect() failed (111: Connection refused) while connecting to upstream, client: 172.30.32.2, server: a0d7b954-nodered, request: "GET /comms HTTP/1.1", upstream: "http://127.0.0.1:46836/comms", host: "homeassistant.local:8123"
> 7 Feb 00:07:38 - [info] Dashboard version 3.1.5 started at /endpoint/ui
> BAIL ON node-red-node-pi-gpio/rpi-gpio
> 7 Feb 00:07:38 - [info] Settings file  : /etc/node-red/config.js
> 7 Feb 00:07:38 - [info] Context store  : 'default' [module=memory]
> 7 Feb 00:07:38 - [info] User directory : /config/node-red/
> 7 Feb 00:07:38 - [warn] Projects disabled : editorTheme.projects.enabled=false
> 7 Feb 00:07:38 - [info] Flows file     : /config/node-red/flows.json
> 7 Feb 00:07:38 - [info] Server now running at http://127.0.0.1:46836/
> 7 Feb 00:07:38 - [info] Starting flows
> 7 Feb 00:07:38 - [info] [OpcUa-Client:53de2f04d7b3701d] 	Local 'own' certificate is NOT used.
> ########################################### OPENSSL ERROR ###########################################
> CWD = .
> Command failed: "openssl" version
> /bin/sh: openssl: not found
> ########################################### OPENSSL ERROR ###########################################
> ########################################### OPENSSL ERROR ###########################################
> CWD = /opt
> Command failed: "openssl" rand  -out "/root/.config/node-red-opcua-nodejs/PKI/random.rnd" -hex 256
> /bin/sh: openssl: not found
> ########################################### OPENSSL ERROR ###########################################`

mikakaraila, 2022-02-06:
As error told, you Docker image doesn't have openssl. Add it to your build.

Normally node-opcua-pki tries to install it during first run but it will fail on Docker.

teun2021, 2022-02-07:
This is a new homeassistant build, using the .vdi from the official homeassistant website. 

openSSL is installed by default, 
[image]

With command 
`openssl req -x509 -newkey rsa:4096 -keyout key.pem -out cert.pem -sha256 -days 365` i just made a new certificate

It still doesnt work.

mikakaraila, 2022-02-07:
Then install node-opcua-pki and test if it can create demo licenses.

https://github.com/node-opcua/node-opcua-pki

teun2021, 2022-02-11:
Thanks! it worked! 
I dont know what i have done as i tried many things. Now its time to play with the nodes's.

closed 2022-02-11 by teun2021