mikakaraila/node-red-contrib-opcua #385: Multiple Subscribe

Eldarion85, 2022-02-07:
Hi, overall it works fine with the "multiple subscription". But I have two questions about this.

1.
If I subscribe to an item in the normal way, information is displayed in the debug. It contains, among other things, the interval I have set.
[image]

If I subscribe to several variables via an array, I can specify an interval, but it's not used. Also, I don't know if I'm specifying this parantally for all items or separately for each item in the payload array.
[image]

2.
The second problem is the same as explained above. I have described it again in my screenshot.
After the first run everything works fine and the right items with the right values are always displayed.

> 1. When I create the subscription with 3 differents items, I receive 3 time the same values of the LAST item of the array used to create the sub.
> 2. When I change the value in the same time for the 3 items, it does the same.
> 3. When I change the value of each item one by one, works fine.

_Originally posted by @Eldarion85 in https://github.com/mikakaraila/node-red-contrib-opcua/issues/367#issuecomment-1029933960_

mikakaraila, 2022-02-07:
Interval is copied from the item object. Subscription has own interval that I suppose is under topic.
Have to check from the code how I implemented it...

wz2b, 2022-02-09:
It might be a nice feature to have you be able to set the interval independently ... for example:

```
msg.payload = [
        { "nodeId": "ns=1;s=ACTUAL_CURRENT",     "type": "Float",   "interval_ms": "1000"},  // proposed
        { "nodeId": "ns=1;s=ACTUAL_VOLTAGE",     "type": "Float",    "interval_ms": "15000"}, // proposed
        ...
}

One thing that's not clear to me is whether or not subscription commands like 'subscribe multiple' are idempotent.  In other words, if you subscribe to the same nodeId more than once, does it pile on redundant subscriptions or is it smart enough to consolidate them.  I started wondering about this when my flows were starting up with an "inject once" trigger to kick things off, and I wasn't able to really find documentation to assure me this whole system is robust - for instance of the OPC/UA device goes away for a period of time then comes back.

wz2b, 2022-02-09:
Another thing I noticed about multiple subscriptions is it doesn't pass "browseName" through like it does if you start with an "OPC Item" and an individual subscription.  That's painful, because you have to construct your own name from the topic.  It seems like the output of a SUBSCRIBE would be consistent no matter how you created the subscription?

mikakaraila, 2022-02-10:
Interval is used only in creation of subscription thus it doesn´t help to set it by each item.
For the second comment as subscription returns only changed values it would require additional mechanism to store msg meta-information and then attach individual nodeId & value return by server to other meta-data.

Please take a look into OPC UA specification:
https://documentation.unified-automation.com/uasdkhp/1.4.1/html/_l2_ua_subscription.html

wz2b, 2022-02-10:
Thanks for the explanation of interval.

I don't want to have to track these friendly names individually, so what I did was looked at how the "OPC/UA Item" node works and I replicated it.  I use a function to create an array of the same object that comes out of "OPC/UA Item" then pass them into a normal "SUBSCRIBE" client request.  What that does is calls subscribe once (separately) for every item in my list.  Then what comes out of that block includes msg.browseName.  So it works great, is compact, and pretty easy to understand, and I don't have to keep track of a friendly name for each point, but I am a little worried about what I'm doing under the hood, performance-wise.  Is it different/worse having 30 individual subscriptions instead of one "multiple" ?

wz2b, 2022-02-10:
> In the case of Aggregate Monitored Items the Sampling Interval denotes the interval for which the aggregates are calculated.

I read that document and I understand it except for one thing: what determines the aggregation method?  min, max, mean, accumulation....

mikakaraila, 2022-02-10:
OPC UA HA Historical Data Access provides aggregate functions. 
I have added those to client action read history.
Node-opcua provides just min,max, ave and interpolative aggregate functions.

Eldarion85, 2022-02-11:
> Interval is copied from the item object. Subscription has own interval that I suppose is under topic. Have to check from the code how I implemented it...

I tested again yesterday with the interval. If I subscribe to an item individually, it works perfectly with the interval. If I set 10000 ms, the server only sends me the updated value every 10000 ms. This also works with smaller times. If I set a smaller interval than I set in the server as the minimum sending-interval, the server will use his own interval.

But I can't set an interval for all items that I subscribe via "multiple subscribe". Regardless of whether I specify the interval for each payload element (nodeId) individually (msg.payload[0].interval; msg.payload[1].intervall) or at a higher level under msg.interval for all items. I always get messages every 1000ms on this way.

Eldarion85, 2022-02-14:
Great, thanks in advance for the optimization!

mikakaraila, 2022-02-16:
OK, I will test and publish this soon.

mikakaraila, 2022-02-16:
New version published.

closed 2022-03-02 by Eldarion85