mikakaraila/node-red-contrib-opcua #386: Re-use self-signed certificate after restart/redeployment with docker

HansDamph, 2022-02-08:
Hi, i am using node-red inside docker container on a linux image.

I use node-red to connect to a KEP OPC-UA server. The server requires to manually "trust" the client. After that, the whole KEP Server need to be restartet. 
This connection works until i deploy a new/other version of my docker container on the same machine. They are not running simultaneously. 
Is it possible to copy or safe and reuse the configuration so that after deployment of my container i dont neet to trust new certififcate of node-red?

This would make the handling much more easy. Thank you in advance!

mikakaraila, 2022-02-08:
Copy PKI folder from the old machine to safe place after new version installation copy it back. Simple way to use same certificates.

HansDamph, 2022-02-09:
Thanks mika! Found PKI in docker at /usr/src/node-red/.config/node-red-opcua-nodejs/PKI
I mounted it as volume to keep it after deployment.
But thats just 1/2 of the solution when using docker. I had to give the container a fixed hostname, otherwise it wont work which makes sense but someone has to keep in mind.

closed 2022-02-09 by HansDamph