mikakaraila/node-red-contrib-opcua #387: opcua client can be automatically reconnected

leexuGit, 2022-02-10:
When my process OPC-UA client has an error, my task stops and I want to be able to automatically re-establish the connection，are there any good practices.
[image]

mikakaraila, 2022-02-10:
It should re-connect automatically, but as error seems to be something else than network break or server re-start it cannot use backoff / reconnect mechanism.

If you use certificates then check that server trusts your client certificate.

leexuGit, 2022-02-11:
I did not use the certificate. Under what circumstances will backoff occur? Can it try to reconnect after backoff occurs? 
[image]

mikakaraila, 2022-02-11:
Client must connect once to server then backoff will be available.

leexuGit, 2022-02-17:
My error log, what do I do with it

17 Feb 08:23:02 - [info] 

17 Feb 08:23:02 - [info] Node-RED version: v1.3.2
17 Feb 08:23:02 - [info] Node.js  version: v12.22.1
17 Feb 08:23:02 - [info] Linux 3.10.0-957.el7.x86_64 x64 LE
17 Feb 08:23:02 - [info] Palette editor disabled : user settings
17 Feb 08:23:02 - [info] Loading palette nodes
17 Feb 08:23:03 - [info] Settings file  : /home/collect/collector/.node-red/settings.js
17 Feb 08:23:03 - [info] Context store  : 'default' [module=memory]
17 Feb 08:23:03 - [info] User directory : /home/collect/collector/.node-red
17 Feb 08:23:03 - [warn] Projects disabled : editorTheme.projects.enabled=false
17 Feb 08:23:03 - [info] Flows file     : /home/collect/collector/.node-red/flows_SJPT.json
17 Feb 08:23:03 - [info] Server now running at http://127.0.0.1:1880/
17 Feb 08:23:03 - [warn] 

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

17 Feb 08:23:03 - [info] Starting flows
17 Feb 08:23:03 - [info] [OpcUa-Client:8b6c3be.d9812c8] 	Local 'own' certificate is NOT used.
17 Feb 08:23:03 - [info] [OpcUa-Client:156557a2.5def48] 	Local 'own' certificate is NOT used.
17 Feb 08:23:03 - [info] [OpcUa-Client:7c47f7d4.ccec38] 	Local 'own' certificate is NOT used.
17 Feb 08:23:03 - [info] [OpcUa-Client:18d442dd.1dccad] 	Local 'own' certificate is NOT used.
17 Feb 08:23:03 - [info] [OpcUa-Client:11c13776.5651b9] 	Local 'own' certificate is NOT used.
17 Feb 08:23:03 - [info] [OpcUa-Client:3d49cd58.55b082] 	Local 'own' certificate is NOT used.
17 Feb 08:23:03 - [info] [OpcUa-Client:ecb896f0.dc5398] 	Local 'own' certificate is NOT used.
17 Feb 08:23:03 - [info] Started flows
17 Feb 08:23:03 - [info] [mqtt-broker:iotdb] Connected to broker: mqtt://10.10.10.24:1883
00:41:03.874Z :client_session_impl           :1493  [NODE-OPCUA-W21] Pending transactions:  ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest
00:41:03.875Z :client_session_impl           :1499  [NODE-OPCUA-W22] Warning : your opcua client is sending multiple requests simultaneously to the server ReadRequest 
 ...                                                  please fix your application code
Client node error on:  error: [" error at active reading: _socket has been disconnected by third party"]
17 Feb 08:41:25 - [error] [OpcUa-Client:8b6c3be.d9812c8] Client node error on:  error: [" error at active reading: _socket has been disconnected by third party"]
dispose when pendingTransactions is not empty 
00:41:25.259Z :client_session_keepalive_manag:145    warning : ClientSessionKeepAliveManager#ping_server  Session has been closed and should not be used to perform a transaction anymore
Client node error on:  error: [" error at active reading: Session has been closed and should not be used to perform a transaction anymore"]
17 Feb 08:41:25 - [error] [OpcUa-Client:8b6c3be.d9812c8] Client node error on:  error: [" error at active reading: Session has been closed and should not be used to perform a transaction anymore"]
(node:26974) UnhandledPromiseRejectionWarning: Error: Already disposing
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-pki/lib/pki/certificate_manager.ts:602:19)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:31:71
    at new Promise (<anonymous>)
    at __awaiter (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:27:12)
    at OPCUACertificateManager.dispose (/home/collect/collector/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:462:16)
    at OPCUACertificateManager.<anonymous> (/home/collect/collector/node_modules/node-opcua-certificate-manager/source/certificate_manager.ts:115:32)
    at Generator.next (<anonymous>)
    at /home/collect/collector/node_modules/node-opcua-certificate-manager/dist/certificate_manager.js:8:71
    at new Promise (<anonymous>)
(node:26974) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:26974) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Client node error on:  error: [" error at active reading: _socket has been disconnected by third party"]
17 Feb 08:41:25 - [error] [OpcUa-Client:156557a2.5def48] Client node error on:  error: [" error at active reading: _socket has been disconnected by third party"]
dispose when pendingTransactions is not empty 
[... cut ...]
 8: 0xd65a85 v8::internal::Heap::CollectGarbage(v8::internal::AllocationSpace, v8::internal::GarbageCollectionReason, v8::GCCallbackFlags) [node-red]
 9: 0xd6853c v8::internal::Heap::AllocateRawWithRetryOrFail(int, v8::internal::AllocationType, v8::internal::AllocationOrigin, v8::internal::AllocationAlignment) [node-red]
10: 0xd2ef5b v8::internal::Factory::NewFillerObject(int, bool, v8::internal::AllocationType, v8::internal::AllocationOrigin) [node-red]
11: 0x107158e v8::internal::Runtime_AllocateInYoungGeneration(int, unsigned long*, v8::internal::Isolate*) [node-red]
12: 0x140de99  [node-red]

leexuGit, 2022-02-17:
<--- Last few GCs --->

[26974:0x499b9b0]  1828331 ms: Mark-sweep 2042.5 (2053.1) -> 2041.3 (2050.6) MB, 1924.4 / 0.0 ms  (+ 4.0 ms in 3 steps since start of marking, biggest step 3.0 ms, walltime since start of marking 1940 ms) (average mu = 0.125, current mu = 0.051) allocatio[26974:0x499b9b0]  1828389 ms: Scavenge 2042.0 (2050.6) -> 2041.4 (2050.8) MB, 11.4 / 0.0 ms  (average mu = 0.125, current mu = 0.051) allocation failure 

<--- JS stacktrace --->

==== JS stack trace =========================================

    0: ExitFrame [pc: 0x140de99]
    1: StubFrame [pc: 0x138f574]
    2: StubFrame [pc: 0x1477583]
Security context: 0x04e7b7a008d1 <JSObject>
    3: /* anonymous */(aka /* anonymous */) [0x357a234426b1] [/home/collect/collector/node_modules/node-opcua-secure-channel/node_modules/async/dist/async.js:2948] [bytecode=0x1d8aa1b056b1 offset=26](this=0x2127ee5404b1 <undefined>,0x357a23442521 <JSFunction (sfi = 0x3f870b1aa931)>,1,0x357a23442b61...

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
 1: 0xa1a640 node::Abort() [node-red]
 2: 0xa1aa4c node::OnFatalError(char const*, char const*) [node-red]
 3: 0xb9a68e v8::Utils::ReportOOMFailure(v8::internal::Isolate*, char const*, bool) [node-red]
 4: 0xb9aa09 v8::internal::V8::FatalProcessOutOfMemory(v8::internal::Isolate*, char const*, bool) [node-red]
 5: 0xd57c85  [node-red]
 6: 0xd58316 v8::internal::Heap::RecomputeLimits(v8::internal::GarbageCollector) [node-red]
 7: 0xd64bd5 v8::internal::Heap::PerformGarbageCollection(v8::internal::GarbageCollector, v8::GCCallbackFlags) [node-red]
 8: 0xd65a85 v8::internal::Heap::CollectGarbage(v8::internal::AllocationSpace, v8::internal::GarbageCollectionReason, v8::GCCallbackFlags) [node-red]
 9: 0xd6853c v8::internal::Heap::AllocateRawWithRetryOrFail(int, v8::internal::AllocationType, v8::internal::AllocationOrigin, v8::internal::AllocationAlignment) [node-red]
10: 0xd2ef5b v8::internal::Factory::NewFillerObject(int, bool, v8::internal::AllocationType, v8::internal::AllocationOrigin) [node-red]
11: 0x107158e v8::internal::Runtime_AllocateInYoungGeneration(int, unsigned long*, v8::internal::Isolate*) [node-red]
12: 0x140de99  [node-red]

mikakaraila, 2022-02-17:
Your client cannot connect to server, one reason could be that client is not trusted on server side.

Test by using just client node, it should get session active as it connects to server:
[image]

leexuGit, 2022-02-18:
I read it every 10 seconds. After a period of time, a connection error will occur, which will cause memory overflow later. In the case of anonymous connection, how is the client certificate generated? Or is there any certificate policy configuration on the server side?

mikakaraila, 2022-02-18:
Is client node status "session active"?
If not it is not connected to server and it will try to connect and it seems to cause crash.
And as I wrote reason can be that server has not trusted to client certificate.
Can you check these?

mikakaraila, 2022-08-23:
No comments for a long time... I will close this one.
Certificate manager normally takes care of server & client certificates.

closed 2022-08-23 by mikakaraila