mikakaraila/node-red-contrib-opcua #388: Node-red crashing on Deploy, Suspect OPC UA Certificate manager

AutoTwisty, 2022-02-10:
I´m having issues with node-red crashing, but only on deploy. It otherwise functions according to spec. On crash node- red reports "Already disposing" followed by pointers to different parts, many associated with the OPCUA Certificate manager.
Any ideas on what the cause might be and how I fix this?
This is the full error log:

`10 Feb 08:50:50 - [info] Stopping flows
10 Feb 08:50:50 - [info] Stopped flows
10 Feb 08:50:50 - [red] Uncaught Exception:
10 Feb 08:50:50 - [error] Error: Already disposing
    at OPCUACertificateManager.<anonymous> (C:\Users\Marcus Hallberg\.node-red\node_modules\node-opcua-pki\dist\pki\certificate_manager.js:464:23)
    at Generator.next (<anonymous>)
    at C:\Users\Marcus Hallberg\.node-red\node_modules\node-opcua-pki\dist\pki\certificate_manager.js:31:71
    at new Promise (<anonymous>)
    at __awaiter (C:\Users\Marcus Hallberg\.node-red\node_modules\node-opcua-pki\dist\pki\certificate_manager.js:27:12)
    at OPCUACertificateManager.dispose (C:\Users\Marcus Hallberg\.node-red\node_modules\node-opcua-pki\dist\pki\certificate_manager.js:462:16)
    at OPCUACertificateManager.<anonymous> (C:\Users\Marcus Hallberg\.node-red\node_modules\node-opcua-certificate-manager\dist\certificate_manager.js:58:38)
    at Generator.next (<anonymous>)
    at C:\Users\Marcus Hallberg\.node-red\node_modules\node-opcua-certificate-manager\dist\certificate_manager.js:8:71
    at new Promise (<anonymous>)`

mikakaraila, 2022-02-10:
Do you have only client node(s) in the flow? Or server node(s)?

Just to get idea where error could be... instance of Certificate Manager is tried to release (dispose) and it seems to fail.

Nothing related to node-red-contrib-opcua?

AutoTwisty, 2022-02-10:
I have several clients in the flow, most subscribe but some write. Currently they are all connecting to the same server. No server nodes in the flow.
The crash happens on all types of deploy, as well as on flow restart.
What might be keeping the Certificate Manager tied up and unable to release? Something on the other end (server I´m connecting to)?

mikakaraila, 2022-02-10:
OK, then problem is global certificate manager as you have multiple client nodes.
You can try to set for each client own PKI store folder then each client will have own certificate manager.

AutoTwisty, 2022-02-10:
Also, does every PKI folder need to have the same content as the original pki at " .node-red\node_modules\node-opcua-pki\dist\pki " ?

mikakaraila, 2022-02-10:
OK, I will check global certificate manager and modify code for deploy case.

cedenoVF, 2022-10-12:
Hello @mikakaraila I am getting the same issue, though this could be solved in my windows PC, the flow could be deployed in another environment, where we dont access to the folder directory.

Is there any other solution? 

In general, I do need a client to "READ" other to "WRITE" and another to "SUBSCRIBE".

mikakaraila, 2022-10-12:
First check that openssl is successfully installed to computer. It will be called during installation... node-opcua => node-opcua-pki

mikakaraila, 2022-10-12:
Aah, there was in between error in initialize that variable. I fixed it some time ago... sorry.

Mulberry26, 2023-01-22:
Sry, I am not particularly understand, did you mean that I should copy everything at "C:\Users\InfoLabor\.node-red\node_modules\node-opcua-pki\dist\pki" to the multiple folder at "C:\Users\InfoLabor\AppData\Roaming\node-red-opcua-nodejs"?