mikakaraila/node-red-contrib-opcua #391: Batch read tags from one UA server and write to another UA server

ryangu18, 2022-02-18:
Thank you so much for making such a powerful module. I'm already learning to use it.
This is an example of my practice, read data from the source UA server, and then write to the new UA server, it has been successful,
[image]

However,  how to read data from the UA server in batches, and then write to another UA server? Maybe a few hundred or more. Can you give me a little help?
I've reviewed the related issue in the question list, but I still haven't found an answer. Maybe a small example of yours will help me a lot

Thanks in advance!

mikakaraila, 2022-02-18:
Best way is to subscribe items and then write them to another server as they change (fast & scalable).

The other way is to use read multiple and write multiple, but this will cause load to both servers.

mikakaraila, 2023-02-28:
Not active anymore.

closed 2023-02-28 by mikakaraila