mikakaraila/node-red-contrib-opcua #392: OPCUA client error - "keepAliveManger" and "client secure channel layer" errors

FT2work, 2022-02-21:
Hi all,
I'm working on a Windows 10 based system with the following installations:
- node red version 2.1.1
- node js version 14.18.1
I'm facing an error everytime a deployment/app startup is performed related to an OPCUA client.
I'm working with node-red-contrib-opcua module, version 0.2.258, to have a server/client architecture.
You can have a look to the error in the attached image.
Can anyone help me?

Thank you very much.

[image]

FT2work, 2022-02-21:
I have a single server node, using during namespace initialization.
Then I have 2 clients node, one for read variables and one for write variable.
I leave blank the field "PKI certificate folder" in both cases.

mikakaraila, 2022-02-22:
What you do before errors will occur?
How to repeat this?

FT2work, 2022-02-22:
Hi @mikakaraila ,
sincerely as I wrote in my first post, it's enough to deploy a modification done on the flow (also a stupid modification as a node movement on the flow editor, a code modification is not necessary to generate the error).
Moreover, at every node-red start-up, the error comes out.
Anyway, the OPCUA server/clients are working properly, it seems.
Quite odd thing!

mikakaraila, 2022-02-22:
Cannot repeat problem, are you using Windows or Linux?
It could be that Certificate Manager cannot access PKI folder. Can be caused by access rights...

FT2work, 2022-02-22:
I'm working on a Windows 10 system.
The situation is quite strange, because I tried to deploy the flow on different machines, where different version of node-red/nodeJS are installed.
Only in one of them the error appears (node-red 2.1.1 and node-js 14.18.1)...
As I wrote, I leave the PKI certificates folder field empty in the server/client nodes configuration page

mikakaraila, 2022-02-22:
There can be multiple folder for the PKI, default is config.

C:\Users\karaimi\AppData\Roaming\node-red-opcua-nodejs
[image]

And some old/not used ones, I have been using and testing:
[image]

mikakaraila, 2022-02-22:
My versions:
C:\Users\karaimi\.node-red>node-red -?
Node-RED v2.1.4
Usage: node-red [-v] [-?] [--settings settings.js] [--userDir DIR]
                [--port PORT] [--title TITLE] [--safe] [flows.json]
       node-red admin <command> [args] [-?] [--userDir DIR] [--json]

Options:
  -p, --port     PORT  port to listen on
  -s, --settings FILE  use specified settings file
      --title    TITLE process window title
  -u, --userDir  DIR   use specified user directory
  -v, --verbose        enable verbose output
      --safe           enable safe mode
  -D, --define   X=Y   overwrite value in settings file
  -?, --help           show this help
  admin <command>      run an admin command

Documentation can be found at http://nodered.org

C:\Users\karaimi\.node-red>node --version
v12.16.2

FT2work, 2022-02-22:
Hi @mikakaraila ,
here below a screenshot about my version.
Do you need something more?
[image]

FT2work, 2022-02-25:
Hi @mikakaraila ,
this is what I catched using verbose option...
[image]

mikakaraila, 2022-02-25:
Ok now I know. Error comes from the new variable that has dimensions 0.
Etienne added check that will prevent creating illegal variables.

mikakaraila, 2022-02-28:
This one is working, how you try to create new array to server?
Flow inject node, notice dimension is given as [5]
[image]

UaExpert:
[image]

mikakaraila, 2022-03-01:
Can you provide some small example what you are trying to do?
AddVariable does not check all parameters and there is something wrong...

mikakaraila, 2022-03-02:
Can you test more? I mean will it crash on first variable or later?
What variable types are you using? It could be typo or not supported variable type.

Or will server crash later. Can you access address space if there is not write to variables?

mikakaraila, 2022-03-03:
Can you do more testing, I mean what will trigger error?
Write a value to variable?

FT2work, 2022-03-04:
I'm not sure I can test more...the project is on stand-by and I don't know when it will be resume

mikakaraila, 2022-03-06:
If namespace/nodeId is wrong then I got this message (just merged parameter check into the server):
[image]

mikakaraila, 2022-03-08:
Updated parameter check version available, please test.

closed 2022-04-04 by mikakaraila