mikakaraila/node-red-contrib-opcua #394: Question: Method support in OPC UA server

ste-hoe, 2022-03-01:
Hi @mikakaraila, 

does your OPC UA server support methods? I could not find a server command to add methods, but I loaded a custom xml containing a method. 

This method is correctly added to the OPC UA server, but when calling it in UaExpert I get "BadInternalError" and no activity in NodeRED.

Any hints?

mikakaraila, 2022-03-01:
No I have not added it, it will require some code and then bindMethod to nodeId.

These are supported actions with server:
[image]

ste-hoe, 2022-03-01:
Do you have an idea, how I could realize a method call in NodeRED? I am struggeling on how to implement some kind of callback in NodeRED?

If I had an idea, I could try to implement something...

mikakaraila, 2022-03-01:
This will need new node and there is still some open issues. 
I will do this after I have finished bind variables for the load address space from the xml file.
How much you have experience for creating node-red nodes?

mikakaraila, 2022-03-02:
Ok, got it implemented. You can use topic to find method and function node contains methodFunc.

[image]

ste-hoe, 2022-03-02:
Wow, you are fast!!

Can you provide your solution? Maybe on a branch?

mikakaraila, 2022-03-02:
I will publish it today or tomorrow.
Have to finish bind variables then it will be good.

mikakaraila, 2022-03-03:
Otherwise everything is ready, but I expect you want to use node-opcua variables etc. then you have to use this:
https://nodered.org/docs/user-guide/writing-functions

At the end:
[image]

mikakaraila, 2022-03-03:
I just published new version.

Here updated example:

[image]

[image]

mikakaraila, 2022-03-03:
You can test with ResendData method before and after bindMethod is injected:

[image]

[image]

svkers, 2022-03-09:
I successfully tested your sample server method "ResendData". That works!
Does it also support server methods with output parameters, e.g. UInt32? I tried the following but it didn't work:

```
global.get("ua");
msg.code = async function methodFunc(inputArguments, context) {
    console.log("Method Input arguments: " + JSON.stringify(inputArguments));
    return { 
        outputArguments: [{
            name: "Arg1",
            dataType: UInt32,
            value: "1234"
        }],
        
        statusCode: ua.StatusCodes.Good
        
    };
};
return msg;
```

Do you have any hint for me? Thanks a lot! 👍

mikakaraila, 2022-03-09:
I will make another example...

svkers, 2022-03-10:
Hi Mika, did you already had some time for the additional example that includes server methods with output parameters? Thank you very much for your effort, support and time! Best regards, Sven

mikakaraila, 2022-03-10:
Not yet, real work at the office... should not be a big work. I will try to look at home.

mikakaraila, 2022-03-10:
Original bark method:
[image]

After bindMethod:
[image]

Inject with correct nodeId:
[image]

Function node with needed code:

var ua = global.get("ua");
msg.code = async function methodFunc(inputArguments, context, callback) {
    console.log("Method Input arguments: " + JSON.stringify(inputArguments));
                                
    var nbBarks = inputArguments[0].value;
    var volume = inputArguments[1].value;

    console.log("Hello World ! I will bark ", nbBarks, " times");
    console.log("the requested volume is ", volume, "");
    var sound_volume = new Array(volume).join("*");

    var barks = [];
    for (var i = 0; i < nbBarks; i++) {
        barks.push("Hau hau" + sound_volume);
    }

    var callMethodResult = {
        statusCode: ua.StatusCodes.Good,
        outputArguments: [{
            dataType: ua.DataType.String,
            arrayType: ua.VariantArrayType.Array,
            value: barks
        }]
    };
    callback(null, callMethodResult);
};

return msg;

svkers, 2022-03-14:
This doesn't work for me. When I try it, I get the following exception:

```
14 Mar 00:52:25 - [red] Uncaught Exception:
14 Mar 00:52:25 - [error] UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "TypeError: Cannot read properties of undefined (reading 'StatusCodes')".
```

The error gets caused by the line:

```
var callMethodResult = {
        statusCode: ua.StatusCodes.Good,
        outputArguments: [{
            dataType: ua.DataType.String,
            arrayType: ua.VariantArrayType.Array,
            value: barks
        }]
    };
```

ua is undefined, thus, the above access caused an unhandled exception. I thought that the ua global context gets set by the node-red-contrib-opcua? Did I miss something?

Thanks again!

mikakaraila, 2022-03-14:
Just edit your settings.js:

[image]

svkers, 2022-03-14:
Thanks that solved this issue! Now it works! Great! :)

May I ask you another question that is related to the async server method function provided to node-opcua and NodeRed itself? Is it possible to trigger/execute another node from within the provided async server method?

mikakaraila, 2022-03-14:
Yes, just use method node. Fill in needed objectId/methodId and parameters.

svkers, 2022-03-14:
Hmm, not sure if I understood your proposal. What I am trying to achieve is to query some data points in the async UA server method from a 3rd party NodeRed node (e.g. influxdb) and return those values to the UA client that called the method.

Normally, I therefore create another flow that queries the data from influxdb by using a flow and the corresponding nodes. Thus, I am thinking how to call another NodeRed flows provide this flow some input values and obtain the queried values by this flow that are then returned within the async UA server method to the UA client. :)

mikakaraila, 2022-03-14:
Method node:
[image]

svkers, 2022-03-17:
Thanks again, I think we can close this issue. 👌