mikakaraila/node-red-contrib-opcua #396: Client Method Call: Problems since connection is closed after every call

HansDamph, 2022-03-03:
I am reading screens from a machine hmi using opc ua method.

Since the connection is closed after every call, i get an error from machines opc-ua server after aproximately 10 - 15 calls. I get this error message:

"
BadDataUnavailable: {
        name: "BadDataUnavailable",
        value: 0x809e0000,
        description: "Expected data is unavailable for the requested time range due to an un-mounted volume, an off-line archive or tape, or similar reason for temporary unavailability."
"

When i want to get the next screent the node stays at "executing method" forever. 
Funny is, that she OS of the machine crashed aswell and only reboot helps.

I didnt had this problem with older version of contrib-opc-ua. Can you make an option for the method if you want to keep connection alive or connect / disconnet for every call?
Thank you in advance!

I am sure the problem is with the machine and not contrib-opc-ua, but I cannot change or view the machines opc ua server programming ;-)

[image]

_Originally posted by @mikakaraila in https://github.com/mikakaraila/node-red-contrib-opcua/issues/365#issuecomment-991694789_

mikakaraila, 2022-03-03:
I fixed two possible cases: close session and disconnect client after method is executed.
New version is published, but test also with UaExpert!

HansDamph, 2022-03-10:
Sorry not yet. The operator of this test machine is earlier than expected on maternity leave. It will take some time for sure... But thank you very much for your helf. I will check asap.

mikakaraila, 2022-04-06:
Stale? Closed as nothing reported.

closed 2022-04-06 by mikakaraila