mikakaraila/node-red-contrib-opcua #398: Monitoring items on remote server

pieterlfm, 2022-03-07:
Hi,

I am trying to monitor different topics with different sampling intervals on a remote OPC UA server. Using the SUBSCRIBE action of the OPC UA client node, I can successfully subscribe and monitor a topic from the server. However, after injecting a monitoring action in order to set the sampling interval (using the values below), no new monitored item seems to be added to the subscription on the server side and there is no output for this topic from the client node.

[image]

mikakaraila, 2022-03-07:
NodeId i=2257 is start time it doesn´t change.

pieterlfm, 2022-03-07:
Indeed. I started testing with i=2258 (CurrentTime) to reliably test a change in sampling interval - but no output for that topic either when using monitor action. I switched to 2257 for testing a different topic, but also there I don't get the (one time) value at subscription time (as is the case when using the SUBSCRIBE action), nor does the number of monitored items increase at the server side.

mikakaraila, 2022-03-07:
DateTime is not good nodeId type to use percent or absolute deadband. 
Test with counter or some other DataItem that has EURange. 
Percent will not work if the nodeId doesn´t have EURange. So it is limited to "DataItems" that contain range so code can calculate percent from min - max.

pieterlfm, 2022-03-08:
Yes, thank you very much for the explanation!

The monitor action for other (Int-type) topics seems to work (I get a value at subscription time) - however, I still need to test further with different sampling intervals during periods the values are changing (this is only sporadically the case for this specific server).

closed 2023-06-28 by mikakaraila