mikakaraila/node-red-contrib-opcua #399: Clean up logging to use logging level rather than `--verbose` flag

kcamenzind, 2022-03-07:
I'm interested in silencing most of the logs coming the opcua nodes, and it looks like the logging is currently using the property `RED.settings.verbose` to whether to run both `console.log` and `node.log`. It looks like Node-RED is migrating away from the verbose flag, since introducing the settings for logging level (ie debug vs info vs warn vs error - for example explained [here](https://discourse.nodered.org/t/how-to-change-the-log-level-of-the-exec-debug-nodes/55183/15)). 

How would you feel about the nodes using `node.log`, `node.warn`, etc. instead of the `verbose_log` function (that is only all or nothing)? My thought would be to remove any use of console.log and then to migrate the calls to `verbose_log` to the appropriate logging level. Specifically, I'd like to have the large paragraphs for every tag write to only be logged at a very high verbosity level, like `node.debug`, since I am performing a large number of writes. But, to keep some things like info about the server starting at the info level.

Let me know what you think about this change, and I am happy to implement & open a PR, unless you would like to do it. There may be concerns about backwards compatibility, if someone is relying on the logs and has the `-v` flag set but a lower logging level in the settings for some reason.

mikakaraila, 2022-03-07:
Yes I can modify it easily, it is from year 2015 very initial solution :-)

I have to finish on-going work. Find just something with the new user certificate authentication.

mikakaraila, 2022-03-08:
OK now settings log level can be used to limit messages. 
I changed verbose console output to node.debug the other ones are still on node.warn and node.error

mikakaraila, 2022-03-08:
I published new version. Check it please.

kcamenzind, 2022-03-08:
amazing, thanks! works as expected

closed 2022-03-08 by kcamenzind