mikakaraila/node-red-contrib-opcua #400: Reduce warning verbosity

anxuae, 2022-03-09:
Hi @mikakaraila 

Is it possible to reduce the log verbosity that was introduced in the last version 0.2.262 ?

Currently my dev environment is not connected to PLC, so the logs are continuously flooding my debug console each time I redeploy my flows.

[image]

I thing that only the status displayed on the nodes is enough. 

Thanks

mikakaraila, 2022-03-09:
I already reduced it, did you read my comments.... this one is only valid as client cannot connect to server.
Minor case, but I will check what is in the node status if it is same as in the console I will remove console output.

mikakaraila, 2022-03-09:
It is now in comments, node status has same information. Will come in next version.

closed 2022-03-09 by mikakaraila

mikakaraila, 2022-03-09:
If you need it immediately edit 102-opcuaclient.js file and line 244, put it comments as here:
[image]

anxuae, 2022-03-21:
Hi @mikakaraila 

Thanks for the new version of your lib!
I've still lots of warning in my debug window and changing the logging level in `settings.js` only affect the console, not the window.

[image]

Is there any way that I missed to avoid flooding the debug windows?

mikakaraila, 2022-03-21:
Those are real warnings as you don´t have connection.

anxuae, 2022-03-21:
Ok thanks,

There is also some warning that it my mind are not:
 - SUBSCRIBE is done
 - client is successfully connected
 - cleanup old listener events
 - no session to close

For me it is the normal "start process" when I'm starting Node-RED. And it make me afraid when I see so many warning which just tell that everything it's normal.... Of course I can live with them 😄 .

[image]

mikakaraila, 2022-03-21:
If you want to get rid of verbose/warnings just turn on error level logging.

[image]

anxuae, 2022-03-21:
Not possible, the verbosity in the `settings.js` only impact the console, not the debug window

anxuae, 2022-03-24:
HI @mikakaraila, is there another way to hide those warnings?
Maybe you can change them into "debug" or "trace" level. In that case they will not be displayed in the debug window.

[image]

mikakaraila, 2022-03-24:
You haven´t installed new version?

I changed on client node them to debug messages. As I wrote you can change this on settings.js

  function verbose_log(logMessage) {
      //if (RED.settings.verbose) {
        // console.log(chalk.cyan(logMessage));
        // node.log(logMessage); // settings.js log level info
        **node.debug(logMessage);**
      //}
    }

See & test:
// Configure the logging output
    logging: {
        // Only console logging is currently supported
        console: {
            // Level of logging to be recorded. Options are:
            // fatal - only those errors which make the application unusable should be recorded
            // error - record errors which are deemed fatal for a particular request + fatal errors
            // warn - record problems which are non fatal + errors + fatal errors
            // info - record information about the general running of the application + warn + error + fatal errors
            // debug - record information which is more verbose than info + info + warn + error + fatal errors
            // trace - record very detailed logging + debug + info + warn + error + fatal errors
            // off - turn off all logging (doesn't affect metrics or audit)
            **level: "error",**
            // Whether or not to include metric events in the log output
            metrics: false,
            // Whether or not to include audit events in the log output
            audit: false
        }
    },

mikakaraila, 2022-04-02:
Cannot see you those warnings from here, please add some screenshot so I can comment...

anxuae, 2022-04-06:
Hi @mikakaraila thanks a lot for your last release! The amount of warnings really lower and better to debug my code.