mikakaraila/node-red-contrib-opcua #401: Error registerFactory: EUInformation already registered

efrefrefr, 2022-03-17:
Hi, I need help to solve following error. I'm new in node-red and playing around with node-red-contrib-opcua. Then, the nodes in the package do not start , but in the pallette window, I see that the end-point node is "in use". Also, I found following messages in terminal

Welcome to Node-RED
===================

17 Mar 09:58:55 - [info] Node-RED version: v2.2.2
17 Mar 09:58:55 - [info] Node.js  version: v16.14.0
17 Mar 09:58:55 - [info] Windows_NT 10.0.22000 x64 LE
17 Mar 09:58:56 - [info] Loading palette nodes
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
17 Mar 09:58:57 - [info] Dashboard version 3.1.6 started at /ui
17 Mar 09:58:57 - [warn] ------------------------------------------------------
17 Mar 09:58:57 - [warn] [node-red-contrib-opcua/OpcUa-Item] Error:  registerFactory  : EUInformation already registered17 Mar 09:58:57 - [warn] [node-red-contrib-opcua/OpcUa-Client] Error:  registerFactory  : EUInformation already registered
17 Mar 09:58:57 - [warn] [node-red-contrib-opcua/OpcUa-Browser] Error:  registerFactory  : EUInformation already registered
17 Mar 09:58:57 - [warn] [node-red-contrib-opcua/OpcUa-Server] Error:  registerFactory  : EUInformation already registered
17 Mar 09:58:57 - [warn] [node-red-contrib-opcua/OpcUa-Event] Error:  registerFactory  : EUInformation already registered
17 Mar 09:58:57 - [warn] [node-red-contrib-opcua/OpcUa-Method] Error:  registerFactory  : EUInformation already registered
17 Mar 09:58:57 - [warn] ------------------------------------------------------
17 Mar 09:58:57 - [info] Settings file  : C:\Users\effen\.node-red\settings.js
17 Mar 09:58:57 - [info] Context store  : 'default' [module=memory]
17 Mar 09:58:57 - [info] User directory : \Users\effen\.node-red
17 Mar 09:58:57 - [warn] Projects disabled : editorTheme.projects.enabled=false
17 Mar 09:58:57 - [info] Flows file     : \Users\effen\.node-red\flows.json
17 Mar 09:58:57 - [info] Server now running at http://127.0.0.1:1880/
17 Mar 09:58:57 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

efrefrefr, 2022-03-17:
[image]

I already uninstall node-red-contrib-opcua in the terminal via typing
npm uninstall node-red-contrib-opcua

but the nodes still show up in the pallate menu but they are not found in the nodes panel (left panel of node red window).

Please help

mikakaraila, 2022-03-17:
Do you have example flow how to repeat this?

Endpoint node is configuration node and it is used by client node.

efrefrefr, 2022-03-17:
Thank you for reply.
I found out that the 1/7 nodes, i.e. opcUa-endpoint is always in use even though i don't have it in any flow (sheet).

Then, if i re-install back node-red-contrib-opcua, the error messages appear in the terminal. I copied and pasted from the terminal.

17 Mar 17:30:45 - [info] Starting flows
17 Mar 17:30:45 - [info] Started flows
17 Mar 17:31:28 - [info] Installing module: node-red-contrib-opcua, version: 0.2.262
17 Mar 17:31:36 - [info] Installed module: node-red-contrib-opcua
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
[class EUInformation extends ExtensionObject] {
  possibleFields: [ 'namespaceUri', 'unitId', 'displayName', 'description' ],
  dataTypeNodeId: ExpandedNodeId {
    identifierType: 1,
    value: 887,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultBinary: ExpandedNodeId {
    identifierType: 1,
    value: 889,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultXml: ExpandedNodeId {
    identifierType: 1,
    value: 888,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  },
  encodingDefaultJson: ExpandedNodeId {
    identifierType: 1,
    value: 15376,
    namespace: 0,
    namespaceUri: null,
    serverIndex: 0
  }
}
target namespace = http://opcfoundation.org/UA/
17 Mar 17:31:38 - [info] Added node types:
17 Mar 17:31:38 - [info]  - node-red-contrib-opcua:OpcUa-Item : Error:  registerFactory  : EUInformation already registered
17 Mar 17:31:38 - [info]  - node-red-contrib-opcua:OpcUa-Client : Error:  registerFactory  : EUInformation already registered
17 Mar 17:31:38 - [info]  - node-red-contrib-opcua:OpcUa-Browser : Error:  registerFactory  : EUInformation already registered
17 Mar 17:31:38 - [info]  - node-red-contrib-opcua:OpcUa-Server : Error:  registerFactory  : EUInformation already registered
17 Mar 17:31:38 - [info]  - node-red-contrib-opcua:OpcUa-Endpoint
17 Mar 17:31:38 - [info]  - node-red-contrib-opcua:OpcUa-Event : Error:  registerFactory  : EUInformation already registered
17 Mar 17:31:38 - [info]  - node-red-contrib-opcua:OpcUa-Method : Error:  registerFactory  : EUInformation already registered

efrefrefr, 2022-03-17:
Sir, Is there a way to clean install this opcua package?

efrefrefr, 2022-03-17:
The following flow worked just fine before the error messages appear. But now, it does not work anymore. It says about missing   nodes. 
[image]

efrefrefr, 2022-03-17:
Sorry, the flow is blocked in previous picture. I recapture the screen
[image]

mikakaraila, 2022-03-17:
Did you edit something? Missing node types mean that node-red cannot register nodes.
Normally caused by error in the node javascript or some missing npm package.

I expect you have messed up your environment somehow...

mikakaraila, 2022-03-19:
This seems to be caused by latest node-opcua. You can downgrade it by command:
npm install node-opcua@2.64.1

I fixed my package to install version above.

closed 2022-03-23 by mikakaraila