mikakaraila/node-red-contrib-opcua #402: Question: Support of OPC UA server events, conditions and alarms?

svkers, 2022-03-17:
How to implement and fire OPC UA server events, conditions and alarms? Are there any example flows available? 

Any hints and links are welcome!
Thanks!

mikakaraila, 2022-03-17:
Look examples OPCUA-TEST-NODES.json

mikakaraila, 2022-03-18:
Use this:
[image]

Then you will get:
[image]

svkers, 2022-03-18:
Hmm, as far as I understand the above the flow, it subscribes to a UA server object for subscribing to events generated by the given OPC UA server endpoint, right?

I am looking for a flow to "generate and fire" UA events for a UA server running in NodeRed.

mikakaraila, 2022-03-18:
Same way... event notifies normally server object.

svkers, 2022-03-22:
Sorry, I am still struggling to get your example event flow successfully run. Maybe I miss something or have a wrong notion of OPC UA events from the view point of the server. How does the server generate the event? Is this done automatically by the server (probably right for some event type, e.g. LevelLimit, after successful installation of the event (?)), do I have to implement a OPC UA server method that triggers a event? How to do that? Again, I am sorry, I am lost with OPC UA events and kindly ask your for a more complete example. I am very grateful for your time and support!

mikakaraila, 2022-03-22:
For the server node:
1) AddVariable
2) InstallDiscretAlarm
3)InstallLimitAlarm

[image]

[image]

mikakaraila, 2022-03-22:
Modify these values to generate new events:
[image]

[image]

closed 2023-02-28 by mikakaraila