mikakaraila/node-red-contrib-opcua #403: running node-red-contrib-opcua in Windows issue

dkruger000, 2022-03-18:
Hi,
I have three Windows 10 systems each running Node.js 14.19.0 and Node-RED 2.1.3 that will be used for production release. Whenever I import your OPC-UA server examples (here _OPCUAServer_Flow.json_), Node-RED crashes completely. The said example does work on a Raspberry Pi running Buster with the exact versions of Node.js and Node-RED as on the Windows machines. Attached is the console output for example _OPCUAServer_Flow.json_

[console_output.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/8305891/console_output.txt)

Thank you

mikakaraila, 2022-03-19:
It is related to address space loading:
Please ensure that the required namespace  http://opcfoundation.org/UA/DI/ is loaded firs when loading  http://opcfoundation.org/UA/AutoID/

Please check npm packages first with command on both computers, in the .node-red folder what is installed:
npm list

mikakaraila, 2022-03-19:
This seems to be caused by latest node-opcua, I changed my package to use node-opcua@2.64.1 it will fix this.
You can downgrade by manually installing older node-opcua.

dkruger000, 2022-03-20:
Thank you for the feedback and the fix. I removed the current version of the module and reinstalled 0.2.263. It solved my issue.
Regards

closed 2022-03-20 by mikakaraila