mikakaraila/node-red-contrib-opcua #404: OPC UA Item node functionality problem in Version 2.262

rugee-f, 2022-03-18:
Hi Mika,I  recently updated Node-Red to version 2.2.2 and my OPC UA nodes to version 0.2.262.
I found the OPC UA Item node not working like before (version 0.2.242) 
[image]
Hitting the red marked inject button will cause the warning of the OPC UA Client node and the DateTime value is not written to the server. Other data types seems to be working (I did not test all).
The green marked inject is creating basically the same object in the function node and is writing the correct value to the server.
In the same flow on version 0.2.242 both injects (yellow and red) are working fine. 
[image]
Some functionality must be broken in the new version when writing timestamps to DateTime items.
Best regards
Frank

mikakaraila, 2022-03-19:
Yes, conversion/validation caused this. I added conversion and now it should work.

rugee-f, 2022-03-19:
Hi Mika, many thanks for your really fast response.

closed 2022-03-20 by mikakaraila