mikakaraila/node-red-contrib-opcua #405: Change value of an OPC UA variable on Node-Red OPC-UA Server

wjomo, 2022-03-20:
Hi Mika,
thanks for your great work with the node-red-opcua toolbox.

I am using version 0.2.263 of node-red-contrib-opcua.

Actually, i make my first steps with OPC UA on  Node-Red.

I am trying to set the value of an variable to the OPC-UA server. 
I have loaded an Custom NodeSet with the OPC UA Server, that works fine.
Now i would change the value of an variable in that NodeSet.

With the UA Expert Client the variable looks like this: 
[image]

The server setup looks like this:
[image]

The identifierType of the variable is "Numeric". How can i address this variable in Node-Red and change the value of it?
In the examples i only can find some code for addressing  some variables witch identifierType is "String", for example:
        **{ payload : { "messageType" : "Variable", "namespace" : "1", "variableName": "Counter",  "variableValue": msg.payload }};**

Thank you for your help and have a nice day.

mikakaraila, 2022-03-20:
Just use client node and item node with correct parameters.
Or simply write value with UaExpert.
NOTE: Allow also anonymous user as you have Security mode: None to access server easily as you test it.

wjomo, 2022-03-20:
Hi Mika,
thank you for your very quick answer. The solution with client node an item node works fine.

Before I created my own NodeSet with Siemens SIMOE, I've tried to setup a folder and some variables to the OPC Server with the opcuaCommands from the Node. That works fine. After setup the folder and the variables, I've tried to save the generated address space with the following Inject Node:
[image]

The server node was configured as shown:
[image]

After injecting the save command to the OPC Server, an error message from the server node occurs:
[image]

Can you please tell me, how I've to configure the server node and the inject node for saving/loading the custom nodeset?

Thanks for your professional help.

mikakaraila, 2022-03-20:
There is example in the OPCUA-TEST-NODES.json
BUT we have to wait a little bit as node-opcua latest version contains changes that will require loading nodeset files in correct order.
Etienne is working with this... it will be after the current node-opcua v2.65.1

wjomo, 2022-03-20:
Thanks a lot for that quick answer.

At the example in the OPCUA-TEST-NODES.json there is the same inject node for saving/loading the address space which I tried to use in my flow with that shown error message.

As you said, i think we have to wait a litte bit for the release of node-opcua v2.65.1.

I'm looking forward to it. Do you have an idea, when the now version will be released?

Thank you, and have a nice evening. 

:+1:

wjomo, 2022-03-28:
Hi Mika,
I have one more question regarding changing an value of an OPC UA variable on Node-Red OPC UA Server. 
I have loaded an custom Nodeset to the OPCUA Server. This works fine. The nodeset was created with SiOME XML Editor from Siemens. In UAExpert the variable looks like this:
[image]

If I try to change the value of that OPC variable straight to OPC Server node with following function node, nothing happens:

`msg =   { payload : { "messageType" : "Variable", "namespace" : "5", "variableName": "Objects.Maschine_1.Test_Bool",  "variableValue": msg.payload }};
return msg;`

After that, I tried to write the value of that variable with an OPC Client and an OPC Item, connected to the server, that works.

[image]

I have no idea, why it works with the client and not straight to the server with the code in the fuction node.

I hope you can tell me what I'm doing wrong.

Thank you for your help.

Best regards

mikakaraila, 2022-03-28:
Value must be true / false with the Boolean type.
But as it works with client node, I will take a look later this week.

wjomo, 2022-03-29:
Hi Mika,
thanks for your quick answer and the perfect support 👍

My inject node for the function node, which doesn't write the value to the server, injects true/false with the datatype Boolean.
[image]

Thank you.

Best regards

wjomo, 2022-04-08:
Hello Mika,
I hope you are fine.
Could you already take a look at the topic described above regarding writing an value of a variable straight to the OPC UA Server Node by using a function node?

Thanks for your help.

mikakaraila, 2022-04-09:
Ok, if not during weekend then on Monday...

mikakaraila, 2022-04-10:
OK v0.2.272 is now available. Please test.

wjomo, 2022-04-11:
Hi Mika,
sorry for my late answer.
Now i have updated on v0.2.272. I wasn t able to update the nodes from the pallete manager on Node-Red. Finally i have removed node red contrib opcua completely after that i installed the new version. Node Red succeeded with no errors.

Now i have tried to write the value of an OPC UA variable which was created from my own address space to the OPC UA Server, it doesn t succeed. I dont know what I am doing wrong. If I try to write the value by using an item and an client node, it works.

Code from the function node with which i try to write the variable directly to the server:

`msg =   { payload : { "messageType" : "Variable", "namespace" : "5", "variableName": "Objects.Maschine_1.Test_Bool",  "variableValue": msg.payload }};
        
return msg;`

Debug-node from the fuction node which is connected to the OPC UA Server:

[image]

UAExpert variable:
[image]

I have no idea what i could do next?

Thank you for your help.

Best regards.

mikakaraila, 2022-04-11:
I think it´s still my fault: variable Value in case of Boolean can be 0 or 1 or "true" or "false" but not true or false.
Stupid error in value conversion... I will fix it tomorrow, but you can try those above.

mikakaraila, 2022-04-11:
I fixed it, version v0.2.273 contains now correct handling.

wjomo, 2022-04-11:
Hi Mika thanks for your quick answer. 👍
I have updated the nodes to v0.2.273 but I am still not able to write that value.

I have tried different data types in my inject-nodes

0/1 as number
"true" / "false" as string
true / false as boolean

but no of this combination works.

Writing the variable with item and client node still works.

What do you think, could be something wrong in my configuration or code?

Thanks for your help.

mikakaraila, 2022-04-13:
Now I need your flow that I can check it. In my tests this was working.

mikakaraila, 2022-04-13:
My tests:
[image]

mikakaraila, 2022-04-13:
Are you missing datatype from the msg?
[image]

wjomo, 2022-04-13:
Hi Mika,
thank you for the advice in terms of datatype. In my function node, there is no datatype defined. I will test it this evening.

Have a nice day.

mikakaraila, 2022-04-13:
This explains your problem: Etienne is still fixing nodeset loading:
https://github.com/node-opcua/node-opcua/pulls

There is something to do with the extension object loading. So we have to wait until it is fixed.

mikakaraila, 2022-04-14:
Good news, I got this working. I added extra code to set server value.
I have to do some extra testing but Happy Eastern!

wjomo, 2022-04-14:
Hi Mika,
thank you for that good news. You are really doing a great job with that powerful OPC-UA nodes.
Can I do some tests or must i do an update of the node-red-contrib opcua nodes or change some Code in my flow?

Dont hurry up, enjoy the Eastern Holidays.

Thanks for the perfect support.

I wish you Happy Eastern too.

Best regards

mikakaraila, 2022-04-18:
Back at home and i just published new version, please test again.

wjomo, 2022-04-18:
Hi Mika,
i hope you had some pretty Eastern-Days.
I have tested the new version. Now the writing of an value straight to the server works fine.

You do an really great job. 

Thank you for that perfect support 👍.

Have an nice evening.

closed 2022-04-19 by mikakaraila