mikakaraila/node-red-contrib-opcua #406: DataType Time . BadTypeMismatch error

alesutto, 2022-03-21:
Good morning, 

I am using these nodes to communicate to a Siemens PLC from Node-Red.
I stumbled upon a limitation as I am trying to communicate a value into a variable that is of type 'Time' 

[image]

When downloading the xlm file from the PLC describing the OPCUA structure, the indicated DataType is "TIME"

[image]
 
I tried to define the msg.datatype as "Time", "DataTime", "String" (setting the payload to T#30s) but nothing seemed to work.
All the time the OPC UA Client node gives back a BadTypeMismatch error.

Any help on this?

mikakaraila, 2022-03-21:
It seems to be Siemens own type and not supported as pure OPC UA type. 
You have to convert it inside PLC to Double/Float/UInt32 to handle it as basic OPC UA type.

alesutto, 2022-03-22:
Thanks,

That solved the problem!
communicating the value as UInt and then converting in the PLC did the trick.

Here a link that can help for the conversion in the PLC.
https://support.industry.siemens.com/tf/ww/en/posts/converting-int-to-time-in-tia-portal-v15/183978

closed 2022-03-22 by alesutto