mikakaraila/node-red-contrib-opcua #408: Recursive browsing

anxuae, 2022-03-23:
Hello @mikakaraila 

One question concerning browsing. I it possible to to it recursively on **only for one part** of the tree?

I've tried using a Client node: but all OPCUA nodes are returned even if I'm pointing out a specific node.

[image]

And when I'm using the browser node, it's not recursive and the nodeId (in topic) to start the browsing is not respected.

(still using the XML nodeset that I sent to you)

Here is my flow:

```
[
    {
        "id": "6fd56ed7.b7bff8",
        "type": "inject",
        "z": "861262c1.684ea",
        "name": "",
        "props": [
            {
                "p": "payload.collect",
                "v": "true",
                "vt": "bool"
            },
            {
                "p": "payload.actiontype",
                "v": "browse",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payloadType": "str",
        "x": 170,
        "y": 1020,
        "wires": [
            [
                "ed8a0456.281de8"
            ]
        ]
    },
    {
        "id": "ed8a0456.281de8",
        "type": "OpcUa-Browser",
        "z": "861262c1.684ea",
        "endpoint": "81ceee10.51d4d8",
        "item": "",
        "datatype": "",
        "topic": "ns=6;s=\"DB_0100_HelloWorld\".\"SystemStatus\"",
        "items": [],
        "name": "",
        "x": 430,
        "y": 1020,
        "wires": [
            [
                "921c6e5e.bd2278"
            ]
        ]
    },
    {
        "id": "921c6e5e.bd2278",
        "type": "debug",
        "z": "861262c1.684ea",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "x": 750,
        "y": 960,
        "wires": []
    },
    {
        "id": "d414d20e.e73668",
        "type": "OpcUa-Client",
        "z": "861262c1.684ea",
        "endpoint": "81ceee10.51d4d8",
        "action": "browse",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "folderName4PKI": "",
        "name": "",
        "x": 420,
        "y": 940,
        "wires": [
            [
                "921c6e5e.bd2278"
            ]
        ]
    },
    {
        "id": "39a6265a.03ed5a",
        "type": "inject",
        "z": "861262c1.684ea",
        "name": "",
        "props": [
            {
                "p": "collect",
                "v": "true",
                "vt": "bool"
            },
            {
                "p": "actiontype",
                "v": "browse",
                "vt": "str"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "ns=6;s=\"DB_0100_HelloWorld\".\"SystemStatus\"",
        "payloadType": "str",
        "x": 170,
        "y": 940,
        "wires": [
            [
                "d414d20e.e73668"
            ]
        ]
    },
    {
        "id": "81ceee10.51d4d8",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://localhost:53801/PLC1/",
        "secpol": "None",
        "secmode": "None",
        "none": false,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]
```

mikakaraila, 2022-03-23:
OK, have to check it. At least it should work as before, it gives topic to crawler:

        // Browse from given topic
        const root = msg.topic; // example topic=ns=0;i=85 "ObjectsFolder";
        const rootObjects = await crawler.read(opcua.resolveNodeId(root));

I will take a look...

mikakaraila, 2022-03-23:
There was mistake after refactoring I made now partial fix.
It will need another round later or I have to revert some changes.

mikakaraila, 2022-03-23:
I reverted back to old implementation. NOTE: I added msg.objects to contain actual results.

anxuae, 2022-03-24:
Seems not solved:

[image]

anxuae, 2022-03-24:
For OPCUA Client node it's solved, thanks!

closed 2022-03-28 by mikakaraila