mikakaraila/node-red-contrib-opcua #409: creating methods for opc ua server

KristianSusiak, 2022-03-23:
Hello @mikakaraila , 
[image]

I m doing my master's thesis creating opc ua information model something like in picture above. My question is how can I create methods in my folder or variable. 

thanks for your response

my flow 

`[
    {
        "id": "ef71d461d345abb3",
        "type": "tab",
        "label": "Flow 2",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "e6de48ae13378649",
        "type": "inject",
        "z": "ef71d461d345abb3",
        "name": "addVariable-Speed",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "ns=1;s=Speed;datatype=Float",
        "payload": "{ \"opcuaCommand\": \"addVariable\" }",
        "payloadType": "json",
        "x": 130,
        "y": 120,
        "wires": [
            [
                "663db0b74c79b6e2"
            ]
        ]
    },
    {
        "id": "18849617162ad63c",
        "type": "inject",
        "z": "ef71d461d345abb3",
        "name": "addVariable-Power",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "ns=1;s=Power;datatype=Float",
        "payload": "{ \"opcuaCommand\": \"addVariable\" }",
        "payloadType": "json",
        "x": 130,
        "y": 160,
[... 502 lines cut ...]
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    },
    {
        "id": "85144921.8931b8",
        "type": "ui_group",
        "name": "dp",
        "tab": "6cee128165972b61",
        "order": 1,
        "disp": true,
        "width": "6",
        "collapse": false,
        "className": ""
    },
    {
        "id": "6cee128165972b61",
        "type": "ui_tab",
        "name": "Teplomer",
        "icon": "dashboard",
        "disabled": false,
        "hidden": false
    }
]`

KristianSusiak, 2022-03-24:
I tried this program, but I can't add any methods. Can I ask you if you could do simple flow with object and method inside?

I will be so much grateful. I am really struggling with this.

mikakaraila, 2022-03-24:
Yes, the example flow will add new code and bind it to existing method.

I should add new command like addVariable => addMethod.
It will require some hours to code & test it. I can look next week.

KristianSusiak, 2022-03-24:
Thank you for your time !

mikakaraila, 2022-03-25:
I have to add some parameters + parameter type conversion but in principle seems to work.

[image]

KristianSusiak, 2022-03-27:
Looks good to me! Can you share what u inject as addMethod?

mikakaraila, 2022-03-28:
I will make it complete, I mean add example with bindMethod. But today evening.

mikakaraila, 2022-03-28:
Ok, now this work. You can define your own method and then inject needed code into it with function node.

[image]

KristianSusiak, 2022-03-29:
Thank you very much, can you share your flow to see what u did? and Is it okey with you, if I add you in mine master's thesis as source?

mikakaraila, 2022-03-29:
Just finishing testing, didn't manage to make it yesterday.

Here is my official profile/signature:
[Mika Karaila ]()
Research Director (Dr.Tech.)
Valmet Automation Inc.
+358 40 7612563
[mika.karaila@valmet.com](mailto:mika.karaila@valmet.com)
Lentokentänkatu 11, Tampere FI-33101
FINLAND

mikakaraila, 2022-03-29:
Here is flow, wait new version. I have not yet committed and published it.
[OPCUA-TEST-NODES.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/8370369/OPCUA-TEST-NODES.txt)

KristianSusiak, 2022-03-29:
> Here is flow, wait new version. I have not yet committed and published it. [OPCUA-TEST-NODES.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/8370369/OPCUA-TEST-NODES.txt)

KristianSusiak, 2022-03-29:
[image]

KristianSusiak, 2022-03-29:
[image]

I have this "Server node error on:  error: "Unknown OPC UA Command"" what should I do?

closed 2022-03-30 by mikakaraila