mikakaraila/node-red-contrib-opcua #411: Problems by writing DateTime Timestamps to Server

rugee-f, 2022-03-26:
Hi Mika, 
I am not sure if the writing of DateTime values to an OPC UA server (I tested Siemens and ProsysSimServer) is working since your last adjustment of this. My application is not transfering any DateTime value to any server since the update from 0.2.262 to 0.2.263.

When I send the actual Timestamp, the value "1970-01-01T00:00:00.000Z" is written to the server.
[image]

mikakaraila, 2022-03-28:
Ack, I made one fix that was wrong type conversion. Have to re-test this.

rugee-f, 2022-03-29:
Hi Mika, is this maybe a version problem of your ProsysSimServer? 
I saw that my namespace (ns=5) is different then yours (ns=1).
But I think this must be a general problem, because writing DateTime to the Siemens OPC-UA server is also failing.

mikakaraila, 2022-03-29:
I made fix that will show actual error under client node and as it told me first NodeId not found then the next problem is attribute error with own server and Prosys Simulation server. I will continue late today debugging.

mikakaraila, 2022-03-29:
Finally solved!

rugee-f, 2022-03-29:
Almost there! Sending timestamp format from inject node is OK.
[image]
But sending a Time String still not working!

[image]

[image]

I try to add my flow...

[DateTime_Flows.json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/8374259/DateTime_Flows.json.txt)

mikakaraila, 2022-03-30:
Ack, I made other re-factoring at the same time. Have to check those...

mikakaraila, 2022-03-31:
New version published, please test & close ticket.

rugee-f, 2022-04-01:
Hi Mika, I tested the new version and I have still one issue.
When I am using your OPC UA Item node the write operation is OK! 
[image]

But when I try to use my function node the null is written to the sever. 
[image]

When I inspect the msg objects passed from your OPC UA Item node and my function node, they seem to be identical.
Are you doing something magic in your node?

It would be great if the write procedure without your OPC UA Item node would work too, because I used this function extensively in my project. With the update to 2.267 the old functionality is still broken.

mikakaraila, 2022-04-02:
As I have fixed my code, I will need some example to fix your code.
Just inject node => function node  => client node would be enough to understand what is problem.

mikakaraila, 2022-04-02:
OK this is working now. Look example OPCUA-RIGHTS.json if you want to use same endpoint extend name to select different user credentials.

[image]

rugee-f, 2022-04-02:
Hi Mika, sorry I do not clearly understand. I think I have not a user credential problem.
My problem is, when I look at the message that your OPC UA Item node is putting out, it looks exactly the same like the message my function node is putting out. My function nodes message is causing that a null is written to the server, your node is writing a datetime value. My function nodes code is like: 

var create = [];
create[1] = {
                   "payload": new Date(msg.payload),
                   "topic":'ns=5;s=DateTimeDataItem',
                   "datatype":'DateTime',
                   "browseName": ''
                   };

for (var i = 0; i< create.length; i++) {
      msg = create[i]
      node.send(msg);
      }
return;

(with the create[] array it is easy to write multiple items and I use this in almost all of my write actions to my server)

But I posted my test flows already  in my comment 4 days ago.

rugee-f, 2022-04-02:
Oh, I saw you created a new version - I will test...

rugee-f, 2022-04-02:
Installed 2.268 - result is the same. My function node is causing the OPC UA Client node to write a null to the server.

mikakaraila, 2022-04-02:
Of course, because DateTimeDataItem != DateTime DateTimeDataItem contains also Definition

rugee-f, 2022-04-02:
Exact the same function node was working in your 2.62 version. 
When I compare the message object coming out of your OPC UA Item node I see no difference to the message object coming out from my function node.  So maybe you have a hint for me where I should implement "Definition" in my function block (and why this was not needed in 2.62)? 
I see now message in NR Box:
[image]

and no error in debug pane:
[image]

Maybe you can import my published flow to see exactly what I mean - I use Prosys server in it too

mikakaraila, 2022-04-03:
Actually Etienne added into the node-opcua variable value type check and it has caused this.
So you cannot use normal DateTime? I will take a look how much it will be work to add DataItems (next week, not today).

rugee-f, 2022-04-03:
I use your nodes for passing MQTT messages to OPC UA in my IOT application. I put the different MQTT items in an array and send these arrays with my functionblock to the OPC UA Client node (write). The array contains objects of all type of datatype items (datetime, boolean, real, uint16, uint32, etc).  With the updates since 2.63 the functionality was broken - but only datetime Items were concerned from that issue. I rarely use the single OPC UA Item node - only in case I have e.g. a dashboard action writing a single Item to OPC UA like acknowledge something or start/stop a sequence. Loading recipes or submit MQTT messages to OPC UA I do en block with multiple reads/writes.

mikakaraila, 2022-04-03:
Simple fix inside your function node, use Date.parse() 

msg.payload = Date.parse(new Date(msg.payload));
msg.topic   = "ns=5;s=DateTime"; // WAS DateTimeDataItem 
msg.datatype= "DateTime";
return msg;

mikakaraila, 2022-04-03:
OK, I added all three formats into be conversion functions. Next version will support Integer, Date and String as payload value for the DateTime.

mikakaraila, 2022-04-04:
I tested this yesterday with all three payloads. New version is published.

closed 2022-04-04 by mikakaraila

rugee-f, 2022-04-04:
Hi Mika, many thanks for your great support.