mikakaraila/node-red-contrib-opcua #413: How to set up a read-only account

shArpyYAo, 2022-03-30:
Hi Mika,
I have setup and used the users.json file like this:
[image]
But when I use the visitor account to log in to the client, it can still change the data in the server. 
Isn't the Guest role only read-only?
Thank you for your answer.

shArpyYAo, 2022-03-30:
client：
[image]

mikakaraila, 2022-03-30:
No, user can have role, but access rights are on the address space and also permissions.
There are possibilities to setup those, but it will need some configuration to setup those.

shArpyYAo, 2022-03-30:
Hi Mika,
Thank you for your quick reply.
So how do I set access rights on the address space?

mikakaraila, 2022-03-30:
I need to implement couple of new opcua commands that will manipulate variables... no other way to access those.

shArpyYAo, 2022-03-30:
So now all roles can access and modify the values inside the server?

mikakaraila, 2022-03-30:
It is much more fine graned, look this:

[image]

Role can be used to give different permissions.

shArpyYAo, 2022-03-30:
where did you set this up？

mikakaraila, 2022-03-30:
It is one of our products. It has all features implemented.
I can make small example to node-red-contrib-opcua to set access levels and permissions.

shArpyYAo, 2022-03-30:
That is what I want.
Can you notify me when you finish the example to node-red-contrib-opcua？
Thank you very much mika

mikakaraila, 2022-03-30:
Didn´t have so much time today, but UI part is now done:
[image]

mikakaraila, 2022-03-30:
OK, access level is working, have to sleep and continue tomorrow:
[image]

mikakaraila, 2022-03-31:
It was easy and fun to add. Most of time was writing html part for the node.

shArpyYAo, 2022-03-31:
Hi Mika,
I was delayed by some things so I didn't get back to you in time.
I would like to ask when will the OpcUa-Rights node be updated?
[image]
I can't find it on version 0.2.266.
Thank you Mika.

mikakaraila, 2022-03-31:
Short coffee break:
Easy to setup different permissions for each role:
[image]

And works:
[image]

shArpyYAo, 2022-03-31:
Cool, 
I'm looking forward to it.

mikakaraila, 2022-03-31:
Now new version is available, I added also new OPCUA-RIGHTS.json to examples folder.

shArpyYAo, 2022-04-01:
Hi Mika,
I have used your new version.
I set read and write permissions for the engineer role.
[image]
But it still prompts me that I don't have permission to modify the value.
[image]
[image]
[image]
This is my users.json
[image]
This is my flow:
[flows.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/8395820/flows.zip)

mikakaraila, 2022-04-01:
Check with UaExpert, attributes accessLevel, userAccessLevel and permissions for the your variable.
You can try to change value with UaExpert, use same username & password.

shArpyYAo, 2022-04-01:
This is my variable:
[image]
I can not change value with UaExpert, using the same username & password:
[image]

mikakaraila, 2022-04-01:
OK, I have to check that server will activate role as client creates session. It could be the cause for this.

mikakaraila, 2022-04-01:
There has been change inside node-opcua, 
I fixed getUserRole => getUserRoles but there is something else now inside node-opcua. I have to continue this after work...

shArpyYAo, 2022-04-02:
Ok,
Thank you for your efforts.

mikakaraila, 2022-04-04:
I published new version yesterday and updated also example flow OPCUA-RIGHTS.json

closed 2022-04-04 by mikakaraila

shArpyYAo, 2022-04-06:
Hi Mika,
I have tested it and it is available.
Thank you for your efforts.