mikakaraila/node-red-contrib-opcua #414: Invalid Channel after performing transaction on WriteRequest

anxuae, 2022-03-31:
Hi @mikakaraila ,

Sometime this error appears when I'm trying to write a value using a client. Can you tel me what is the meaning? And what is the reason?
Unfortunately I have no more logs to provide. 

```
Client node error on: Client WRITE error: ["Client WRITE Cannot write value ([true]) to msg.topic:ns=6;s=\"DB_Comms_WCS\".\"OutputRequestStatus\".\"ready_to_receive\" error:Error: Invalid Channel after performing transaction on WriteRequest"]
```

Thanks

mikakaraila, 2022-03-31:
Etienne has fixed something to new node-opcua related to keepalive, it could be fixed to latest version.

mikakaraila, 2022-04-05:
Seems to be related to flow that contains server and client is writing a value.
I expect server is just not ready to set value, building address space or something else.
As in my test next write will succeed.

mikakaraila, 2022-04-05:
I added extra check for session before write.

[image]

mikakaraila, 2022-04-05:
New version published v0.2.270