mikakaraila/node-red-contrib-opcua #416: Write not working from Item Node

matteofranceschini, 2022-04-04:
Hi! Thanks for this great library!
I'm having a (mild) issue that took me some time to debug, when trying to write on a Siemens PLC server via OPC-UA. First of all, the server works because I'm able to write and read with UA-Expert.

I have this simple setup:
[image]

If I set up the inject node with a numeric timestamp and the item node "value" to something (like "byebyeworld"), when I trigger the inject node, nothing happens, not catched even by the catch node.
[image]

On the other hand, if I set up the payload of the inject node with "helloworld", everything works, the inject value gets written. If i set again the value in the item node to "byebyeworld", according to the documentation it shell _dominate_ and overwrite the "helloworld" to be sent. This is not what happens, though: the value sent to the server is still "helloworld", despite setting the value in the item node.
[image]

If you need me to test some other cases, LMK.

mikakaraila, 2022-04-05:
Updated item documentation:
[image]

mikakaraila, 2022-04-05:
New version published v0.2.270

closed 2022-04-05 by mikakaraila