mikakaraila/node-red-contrib-opcua #417: Node-red crach when redeploying flows

anxuae, 2022-04-04:
Using flows with OPCUA server, node-red crash frequently when I'm redeploying my flows (this is the case since ):
Here is the logs I got when it's , but I'm not able to reproduce it systematically.

```
wcsnodered                      | 14:59:03.141Z :ua_variable_impl              :1343  UAVariable#_internal_set_dataValue : no addressSpace ! may be node has already been deleted ?
wcsnodered                      | 4 Apr 14:59:03 - [red] Uncaught Exception:
wcsnodered                      | 4 Apr 14:59:03 - Error: Internal error
wcsnodered                      |     at /data/node_modules/node-opcua-server/dist/server_engine.js:510:23
wcsnodered                      |     at cb (util.js:353:39)
wcsnodered                      |     at process._tickCallback (internal/process/next_tick.js:63:19)
wcsnodered                      | npm ERR! code ELIFECYCLE
wcsnodered                      | npm ERR! errno 1
wcsnodered                      | npm ERR! node-red-docker@1.3.7 start: `node $NODE_OPTIONS node_modules/node-red/red.js $FLOWS "--userDir" "/data"`
wcsnodered                      | npm ERR! Exit status 1
```

mikakaraila, 2022-04-04:
Now we need famous debug log :-)

It seems more to be node-opcua related problem as actual server is not designed to be re-started this way.
UA Variable checks/validates type will match with the value, new feature that could cause this.
Looks some kind of timing problem...

mikakaraila, 2022-04-05:
Can be also related to docker environment... I have tested multiple times without any crash with pure Windows.

mikakaraila, 2022-04-11:
What kind of variables are in your flow or how many?

mikakaraila, 2022-06-29:
No activity / log for further studies => closed.

closed 2022-06-29 by mikakaraila