mikakaraila/node-red-contrib-opcua #418: Node-red crashes after using loadAddressSpace command

shArpyYAo, 2022-04-06:
Hi Mika,
When I reload my data with the loadAddressSpace command, node-red crashes.
Node-red version：v2.2.2
Node-red-contrib-opcua version：v0.2.270
My loadAddressSpace：
[image]
My xml file：
[testsave.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/8425693/testsave.zip)
Node-red logs：
[image]

mikakaraila, 2022-04-06:
There has been changes inside node-opcua. Etienne has refactored ExtensionObject. I have to check what it will need...

mikakaraila, 2022-04-06:
I made a new issue to node-opcua: https://github.com/node-opcua/node-opcua/issues/1132

mikakaraila, 2022-05-06:
Look my comment on this:
https://github.com/mikakaraila/node-red-contrib-opcua/issues/374

It is impossible to load ns=1 into the server address space. So you must save/export namespace that is not ns=1.

kmmena, 2023-01-19:
Hello, I am using the commands "To save and Load the backup with the XML file" but when loading Node-Red it crashes and shows me this error, if you can help me or guide me what is it? Thank you
[image]

mikakaraila, 2023-01-24:
Please test with new v0.2.295 as it uses now node-opcua v2.89 that contains fixes to extension objects.