mikakaraila/node-red-contrib-opcua #420: node OPCUAServer is not working

federicab20, 2022-04-06:
Hello, 
I'm working with library node-red-contrib-opcua but since this morning the Node-red flow has stopped working and I don't understand the problem. The OPCUA server in the 'custom nodeset directory' field presents the directory containing the .xml file that it must read to insert namespace values, but seems that the file is not read because I get ns = -1 instead of ns = 5, do you have any idea what happened? I am sending you the screenshots to understand better.

[image]

the file contains:
[image]

I try to update the library from version 0.2.266 to 0.2.270 but without success.

Thank you

mikakaraila, 2022-04-06:
New node-opcua validates nodeset xml and I expect it is not accepted / there is nothing to import.

mikakaraila, 2022-04-06:
There is also registerNamespace in the opcuaCommands

federicab20, 2022-04-06:
I didn't think about it before, thank you

mikakaraila, 2022-04-06:
Your nodeset is not valid. I added there a bit more definitions like alias and after that it will fail to load.
<?xml version='1.0' encoding='utf-8'?>
<UANodeSet xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd">
  <NamespaceUris>
    <Uri>PLC</Uri>
    <Uri>EQTQ</Uri>
    <Uri>NS</Uri>
    <Uri>MUR</Uri>
  </NamespaceUris>
    <Models>
        <Model ModelUri="urn:LAPTOP-000V8I22:NodeOPCUA-Server" Version="0.0.0" PublicationDate="1900-01-01T00:00:00.000Z">
            <RequiredModel ModelUri="http://opcfoundation.org/UA/" Version="1.04.6" PublicationDate="2020-04-14T00:00:00.000Z"/>
        </Model>
    </Models>
    <Aliases>
        <Alias Alias="ApplicationDescription">i=308</Alias>
        <Alias Alias="ApplicationType">i=307</Alias>
        <Alias Alias="Argument">i=296</Alias>
        <Alias Alias="Boolean">i=1</Alias>
        <Alias Alias="Byte">i=3</Alias>
        <Alias Alias="ByteString">i=15</Alias>
        <Alias Alias="Double">i=11</Alias>
        <Alias Alias="Duration">i=290</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="LocaleId">i=295</Alias>
        <Alias Alias="LocalizedText">i=21</Alias>
        <Alias Alias="MessageSecurityMode">i=302</Alias>
        <Alias Alias="NodeId">i=17</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="ServiceCounterDataType">i=871</Alias>
        <Alias Alias="SessionDiagnosticsDataType">i=865</Alias>
        <Alias Alias="SessionSecurityDiagnosticsDataType">i=868</Alias>
        <Alias Alias="String">i=12</Alias>
        <Alias Alias="SubscriptionDiagnosticsDataType">i=874</Alias>
        <Alias Alias="UInt16">i=5</Alias>
        <Alias Alias="UInt32">i=7</Alias>
        <Alias Alias="UtcTime">i=294</Alias>
    </Aliases>
</UANodeSet>

[image]

mikakaraila, 2022-04-06:
This way it will work:
[image]

[image]

There are commands: getNamespaceIndex and getNamespaces.

federicab20, 2022-04-07:
Thank you very much for you explanation.
Just a notification: my nodeset worked, I had seen examples where it was modified in that way. But that's not a problem, this other solution is fine. Thank you.

closed 2022-04-07 by mikakaraila