mikakaraila/node-red-contrib-opcua #421: Multiple Subscribe outputting incorrect values.

simonbowen, 2022-04-06:
I am using the Multiple Subscribe functionality with the opcua-client node. The subscription seems to be set up but for some reason the output is incorrect. I get a duplicated output for one of the subscribed nodes. 

[image]

[flow.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/8427081/flow.zip)

Am I subscribing to the nodes incorrectly?

mikakaraila, 2022-04-06:
Have you monitored value with UaExpert?
Are both Humidity & Temperature changing or only Temperature?

mikakaraila, 2022-04-06:
Yes, subscription is very effective. Think about case you have hundreds or thousands of variables.
Only few of those will change => only changed values are updated. 
This will save a lot network load and is very effective also for the CPU and memory management.

simonbowen, 2022-04-06:
Thank you. 

Do you have any idea why I may be getting duplicate values from my subscription? In the video you can see I receive an update for the same value twice.

mikakaraila, 2022-04-06:
Have you subscribed it twice? Also timestamp or status code change will activate change => value update.

simonbowen, 2022-04-08:
I am not subscribed to it twice. It appears to publish the temperature twice if there is an update, but the humidity once. 

It's an incredibly simple flow with one instance of the OPCUA Client. 

[simple-flow.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/8452714/simple-flow.zip)

mikakaraila, 2022-04-09:
It is somehow depending on sampling/publishing interval. 
Most probably server will send first buffered values and then value change at the same time with new value.

https://reference.opcfoundation.org/v104/Core/docs/Part4/5.12.1/

You can change 5000ms to lower like 1000ms and then you will values once. 
You can try to use monitoring with deadband or percent if you want to filter values.

simonbowen, 2022-04-09:
That seems like the issue. Thank you for responding and I’ll educate myself on OPCUA bit more.

closed 2022-04-09 by simonbowen