mikakaraila/node-red-contrib-opcua #422: NodeRed Error message while trying to write value to OPCUAserver

federicab20, 2022-04-07:
I have a problem with the library: I installed the last version of  node-red-contrib-opcua and I imported your example to create server and write values [OPCUA-TEST.json](https://github.com/mikakaraila/node-red-contrib-opcua/blob/master/examples/OPCUA-TEST.json). The server runs but when I try to write the variable values, I get the following error message:

[image]

What do it happens?  I was able to write the values without any problem.
Thank you

mikakaraila, 2022-04-07:
By default anonymous cannot write.
Access level / permission will deny it.

mikakaraila, 2022-04-08:
Look examples folder and OPCUA-RIGHTS.json
I will take a look, it should work still without using rights node.

[image]

mikakaraila, 2022-04-08:
Can you attach your flow here, I would like to check it.

mikakaraila, 2022-04-08:
Your server does not accept Anonymous user:
[image]

mikakaraila, 2022-04-08:
Thanks, I found it! My code was not setting correctly access levels in case that rights node was not used, I fixed it.
[image]

federicab20, 2022-04-08:
Perfect, now I can write correctly the variable values. Thank you

closed 2022-04-08 by federicab20