mikakaraila/node-red-contrib-opcua #423: saving addressspace not finding provided filename

JanEggers, 2022-04-08:
im a little confused i clearly provide the filename so i dont know why its not picked up.

[image]

mikakaraila, 2022-04-08:
Sorry for some reason msg must contain also topic. Perhaps my copy&paste have caused this...

```Javascript
 case "saveAddressSpace":
                        if (msg.topic && msg.payload && msg.filename) {
                            // Save current server namespace objects to file
                            const namespace = addressSpace.getOwnNamespace(); // TODO msg.topic to select namespace?
                            const xmlstr = namespace.toNodeset2XML();
                            fs.writeFileSync(msg.filename, xmlstr.toString(), {encoding: "utf8"});
                        }
                        else {
                            verbose_warn("Check msg object, it must contain msg.filename for the address space in XML format!");
                        }
                        break;
```

mikakaraila, 2022-04-08:
Reason is in the comments. I have idea to extend to use msg.topic to select namespace that should be saved.

JanEggers, 2022-04-08:
thx that solved the issue

closed 2022-04-08 by JanEggers