mikakaraila/node-red-contrib-opcua #424: creating a complex folder structure is not working

JanEggers, 2022-04-08:
i tried to create a complex folder structure but im having ouble because the parent ssems wrong.

when I do:

addFolder a
addFolder b

I get:

objects/a/b

I would expect:
objects/a
objects/b

i also tried this but the result is the same

addFolder a
setfolder i=85
addFolder b

mikakaraila, 2022-04-08:
setFolder will remember the current created folder as parent this is by my design.
First create folder and then add variables into it. Makes this way it easy.

If you want to set parent for the folder/variables use setFolder. I expect topic should contain namespace like ns=0;i=85

I should perhaps add there some error checking and warning/error message if node is not found.

JanEggers, 2022-04-08:
seems it was a timing issue.
when adding delays the behavior is as desired.

addFolder a
delay 50 ms
setfolder i=85
addFolder b

i would really like to be able to create folders and variables providing the parent folder explicitly so It does not rely on "hidden" current folder

mikakaraila, 2022-04-08:
I cannot change behavior that I build earlier. It will most probably broke many other cases.

And the current behavior has been documented in the node´s own help:
[image]

closed 2022-04-08 by mikakaraila