mikakaraila/node-red-contrib-opcua #426: Cannot read property ‘schema’ of undefined

matteofranceschini, 2022-04-08:
Hi again,
I was working with an "extension object" and I was getting an error on the debugger
[image]
[image]
the error was on the debugger node that has `[struttura esportazione ` in this picture.
[image]

After some googling, I found out that I wasn't the only one, and that this issue is still _unsolved_ . I'm not an expert in nodered nor OPC-UA, but I actually found a workaround: I noticed that, depending on which node I routed first after "read convogliatore", the error was being triggered by a different node: the debugger, the splitter or the "imposta come letti".
As far as I have understood, the routing order in nodered matters: the first node gets the original message, the others get a copy of it (this is actually not so clear to me, but anyway).
I wanted to split the original message object (an array) in multiple messages, an I started using a standard nodered "splitter node"... and this didn't work. Only the *first* node routed node worked, every time. 
I then created a custom function node, routed it to the OPC-UA node and created two outputs: the first one splits the message payload in multiple messages, the second one just forwards a _hard copy_ of the object (with a parse(stringify()) combo).

This works correctly now.
[image]
[image]

mikakaraila, 2022-04-08:
And... ?? What actions I should take? Add this as an example how to handle ExtensionObject...

matteofranceschini, 2022-04-08:
Sadly, I have no knowledge on how the library works nor I have enough knowledge on how nodered works.

It's the first time in years I've found this issue and as far as my googling concerns, it seems to be linked with this library and specifically with the extension object. 

I just hoped that my workaround could trigger someone with knowledge on a possible solution, which I'm absolutely not able to provide. Some uniformed guesses (based on a very thin knowledge of javascript, nodered and opc) could be:
- that the object provided by the "opc-ua client node" gets deleted or modified after the first time it's being sent, and those who reference it afterwards have just an incomplete or undefined reference
- the object provided by the node has somewhere the information on how the data shall be processed, but it's deleted afterwards and just a plain string (or byte buffer) remains, with no info on how to parse it
- the library triggers a hidden/unknown bug of nodered, on objects and passing object references

I don't think that my workaround is the correct way of handling an ExtensionObject, because if I just have one node after it everyhthing works. So the extensionObject _by itself_ is correctly ridden, parsed and forwarded to the following node. For some reason, though, this does not work if the forwarding should be done _multiple times_ because I have more nodes connected to the output. 

I'll do some other tests, but I think that there **could** be some kind of issue in the way the object passed by the "opc-ua client node" to the other node-red nodes. :)

mikakaraila, 2022-04-08:
Ok, actual extension object has been  a bit more complicated as also the specification is under update by OPC Foundation.
And the implementation has not been complete by node-opcua. 
Etienne has been improving all the time but it needs still some work. I just have to wait and adapt to updates...