mikakaraila/node-red-contrib-opcua #429: Cannot create server using GUID as identifier type in NodeId

anxuae, 2022-04-22:
Hi, 

I've got the following error when I try to create a server using GUID in a NodeId:

```
ns=5;g=00000000-0000-0000-0000-000000000000
```
[image]

Here is a simple example flow:

```
[{"id":"cc5b4f8e.beec88","type":"OpcUa-Server","z":"d3d1ff6.817a48","port":"4841","name":"OPC UA server","endpoint":"","users":"C:\\Users\\SteeveVermeersch\\.node-red\\users.json","nodesetDir":"","folderName4PKI":"","autoAcceptUnknownCertificate":true,"registerToDiscovery":false,"constructDefaultAddressSpace":false,"allowAnonymous":true,"endpointNone":true,"endpointSign":false,"endpointSignEncrypt":false,"endpointBasic128Rsa15":false,"endpointBasic256":false,"endpointBasic256Sha256":false,"maxNodesPerBrowse":"","maxNodesPerHistoryReadData":"","maxNodesPerHistoryReadEvents":"","maxNodesPerHistoryUpdateData":"","maxNodesPerRead":"","maxNodesPerWrite":"","maxNodesPerMethodCall":"","maxNodesPerRegisterNodes":"","maxNodesPerNodeManagement":"","maxMonitoredItemsPerCall":"","maxNodesPerHistoryUpdateEvents":"","maxNodesPerTranslateBrowsePathsToNodeIds":"","x":860,"y":160,"wires":[["9ee96f82.f63b9"]]},{"id":"9ee96f82.f63b9","type":"debug","z":"d3d1ff6.817a48","name":"server_opcua","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1060,"y":160,"wires":[]},{"id":"2399ac0f.04645c","type":"inject","z":"d3d1ff6.817a48","name":"Create OPCUA Tree","props":[],"repeat":"","crontab":"","once":true,"onceDelay":"3","topic":"","x":140,"y":120,"wires":[["ec74f1ef.a5287"]]},{"id":"a8932b0e.084e5","type":"change","z":"d3d1ff6.817a48","name":"add folder - STATION-SIMU","rules":[{"t":"set","p":"payload","pt":"msg","to":"{\"opcuaCommand\":\"addFolder\"}","tot":"json"},{"t":"set","p":"topic","pt":"msg","to":"ns=5;g=00000000-0000-0000-0000-000000000000","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":400,"y":160,"wires":[["cc5b4f8e.beec88","ea760655.5ae6b8"]]},{"id":"ea760655.5ae6b8","type":"change","z":"d3d1ff6.817a48","name":"set folder - STATION-SIMU","rules":[{"t":"set","p":"payload","pt":"msg","to":"{ \"opcuaCommand\": \"setFolder\"}","tot":"json"},{"t":"set","p":"topic","pt":"msg","to":"ns=5;g=00000000-0000-0000-0000-000000000000","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":420,"y":200,"wires":[["cc5b4f8e.beec88","538ed7fc.1c5648"]]},{"id":"538ed7fc.1c5648","type":"change","z":"d3d1ff6.817a48","name":"add folder - General","rules":[{"t":"set","p":"payload","pt":"msg","to":"{ \"opcuaCommand\": \"addFolder\"}","tot":"json"},{"t":"set","p":"topic","pt":"msg","to":"ns=5;g=00000000-0000-0000-0000-000000000000","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":400,"y":240,"wires":[["cc5b4f8e.beec88"]]},{"id":"ec74f1ef.a5287","type":"change","z":"d3d1ff6.817a48","name":"registerNamespace - STATION-SIMU","rules":[{"t":"set","p":"payload","pt":"msg","to":"{\"opcuaCommand\":\"registerNamespace\"}","tot":"json"},{"t":"set","p":"topic","pt":"msg","to":"\"http://www.exotec.com/STATION-SIMU\"","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":430,"y":120,"wires":[["a8932b0e.084e5","cc5b4f8e.beec88"]]}]
```

mikakaraila, 2022-04-23:
Only String and Integer are currently supported as nodeId identifiers.
I can take a look tomorrow / next week.

mikakaraila, 2022-04-25:
Can you give any more context when this will show up? Loading a nodeset or adding a variable?

mikakaraila, 2022-04-25:
I tested with your flow and I removed last addFolder with already existing Guid.

[image]

mikakaraila, 2022-04-25:
Works if addFolder is NOT empty Guid and variable is valid (next free Guid):
[image]

mikakaraila, 2022-04-25:
I fixed browseName & displayName cosmetic issue.

[image]

mikakaraila, 2022-05-04:
If no comments I expect this is OK

closed 2022-05-04 by mikakaraila

anxuae, 2022-05-04:
Yes it's ok, thanks a lot