mikakaraila/node-red-contrib-opcua #430: Problems with loading a Custom NodeSet

wjomo, 2022-04-22:
Hi Mika,
I have got a small Issue with loading a custom Nodeset with the OPC Server.
In the past, I have done some tests with loading an CustomNodeset. At these Tests I loaded 2-3 variables, that works fine.

Now I enlarged the .xml file with about 70 variables of datatype Boolean and Float. When I browse the Custom Namespace with UAExpert, there are only 11 variables shown.

[image]

[image]

[image]

If the setup for the CustomNodeSet is complete, there should be 13 Objects with respectively about 70 variables.

Can you please tell me, what I am doing wrong with my CustomNodeSet? Or is there a limitation of variables?

Thank you verry much for your help.

Best regards.

[OPC_Hydraulikanlagen_210422_v1.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/8542760/OPC_Hydraulikanlagen_210422_v1.zip)
.

mikakaraila, 2022-04-23:
Hmm, perhaps umlauts are not very good to use as nodeId, û and ä could cause something even no warnings in the console.

wjomo, 2022-04-23:
Hi Mika,
thanks for the very quick answer. :+1:

I edited my Custom NodeSet and removed every umlauts and spaces in the variablename. Now it works. I am able to browse all variables.

Is there a limitation of the Quantity for variables in a CustomNodeSet at the server? Do you think it wll be possible to set up about 1000 variables? 

I try to read out 13 S71200 Plcs with an s7-node, after reading I will write the values straight to the OPC Server. What do you think will the best way for writing the values to the OPC Variables. From the S7 node i get one Object for each PLC with the variablename an its value. It looks like this:

'boolean_Anlagenzustand_aus' : true,
'float_Druck' : 254,3,
.....

I am thinking about writing the values with a function node straight to the server, or have you an idea with a better way for writing that values?

Thanks for your help.

mikakaraila, 2022-04-24:
Another good news: new node-opcua v2.68.0 will accept umlauts. File reader was using ascii, not UTF-8.
I haven´t yet updated node-red-contrib-opcua to use it.

For variables you can use this kind of msg object:
[image]

wjomo, 2022-04-25:
Hi Mika,
thank you for that good news about the file reader. This is advantageous for German-language variable names.

What do you think, would it be a better way to load all variables in one OPC UA Server node (about 1000 variables), or would it be better to set up an OPC UA Server for every machine (13 Server Nodes, every in a single flow, with about 70 variables per Server)

At the moment I am testing the workaround with an testsytem which simulates one machine with about 70 variables. I use a fuction node which generates for every variable an message object as shown above to write that value. This message object is send asynchronously within a "for-loop".
 The cycle-time for updating the variables from the PLC is round about 1000ms. How much time do you think, needs the OPC UA Server for writing one value?

I am thinking about, what will happens if 13 machines writes the values to the server at the same time. :thinking:

Thank you for your help.

mikakaraila, 2022-04-25:
I have been running very large OPC UA Servers with more than 10000 variables (not under node-red).
One way to implement simulation is to use timers or timeout that will re-trigger timeout.
This way it will be totally independent and asyncronous.

wjomo, 2022-04-25:
These are very large servers, wow...

What will happens at the OPC Server, if there are too many input objects for writing the values? Is there are FIFO stoarage inside the server, or generates it some lost data? If I setup one single server for all variables, in worst case there would be about 1000 input objects send to the server for writing the values.

Please excuse my possibly stupid questions. I'm relatively new to node red and java script. I usually work with PLC Controllers.

Thank you.

mikakaraila, 2022-04-25:
Cannot say in your case, I have subscribed 10000 variables from the server and response test with UaExpert is ~150ms.
Depends on server CPU core speed (normally not a memory limited).
Are you simulating each variable individually or can you make ring buffer to read & update simulated values from it?
I made one simulator that will first read whole simulation value dataset from file and then loop it to update all variables.

mikakaraila, 2022-04-25:
OK, I will close this one, but you can comment your performance results here later.

closed 2022-04-25 by mikakaraila