mikakaraila/node-red-contrib-opcua #431: How to incorporate the result of method in a variable of OPCUA server

rizwankamleen, 2022-04-27:
Dear Author/concerned community,

I made a method of just the addition of two numbers (see attached pic) but now I have to save the output arguments in a variable of OPCUA server unfortunately while creating the method, in the function block I can not access the result in msg.payload.

It will be a great help if someone can make a flow for such an issue.

Kind regards,
Rizwan
[image]

mikakaraila, 2022-04-27:
Look some closed issues, I remember that I made example couple of months ago.

mikakaraila, 2022-04-27:
Read this one: https://github.com/mikakaraila/node-red-contrib-opcua/issues/394

rizwankamleen, 2022-04-30:
Thanks for your quick responses, basically I have gone through over #394 and my method command is executing properly but my concern was that if it is possible if we can take the output of the method in a separate variable in OPCUA. E.g. we create an addition method of one input and one output. Now I want to use this output as msg.payload so that we can do further operations in other nodes. 

Is it possible? because I try to do some changes in the function block but doesn't get good.

Please check the function also that I make for the method.
[image]

mikakaraila, 2022-05-02:
Simple rules:
1) addMethod will add method to address space with nodeId, input/output arguments are given as interface
2) bindMethod will add actual function to be executed as method is called
    Inside this function you can do what ever you want but it must return status code or method results (output arguments).

There are another examples inside OPCUA-TEST-NODES.json and they calls Prosys UA Simulation server methods.

mikakaraila, 2022-05-02:
Here is code snippet from the method node, it will save result and take it to the output msg:
```javascript
try {
          const result = await node.session.call(callMethodRequest);
          if (diagInfo) {
            verbose_log("Diagn. info: " + JSON.stringify(diagInfo));
          }
          verbose_log("Output args: " + JSON.stringify(msg.outputArguments));
          verbose_log("Results:     " + JSON.stringify(result));
          msg.result = result;
          if (result && result.statusCode === opcua.StatusCodes.Good) {
            var i = 0;
            msg.output = result.outputArguments; // Original outputArguments
            msg.payload = []; // Store values back to array
            if (result.outputArguments.length == 1) {
              verbose_log("Value: " + result.outputArguments[i].value);
              msg.payload = result.outputArguments[0].value; // Return only if one output argument
            } else {
              while (result.outputArguments.length > i) {
                verbose_log("Value[" + i + "]:" + result.outputArguments[i].toString());
                msg.payload.push(result.outputArguments[i].value); // Just copy result value to payload[] array, actual value needed mostly
                i++;
              }
            }
          } else {
            set_node_status_to("error: " + result.statusCode.description)
            node.error("Execute method result, error:" + result.statusCode.description);
            return result.statusCode;
          }
          node.send(msg);
          return opcua.StatusCodes.Good;
        } catch (err) {
          set_node_status_to("Method execution error: " + err)
          node.error("Method execution error: " + err);
          return opcua.StatusCodes.BadMethodInvalid;
        }
      }
```

rizwankamleen, 2022-05-03:
Thanks a lot for the code, i will try it then update you.

mikakaraila, 2023-02-28:
Closed as not active anymore.

closed 2023-02-28 by mikakaraila